bulletphysics/bullet3 #2647: How to use the C API

AlexApps99, 2020-02-29:
It says in the README:
> A new physics-engine agnostic C-API is created, see examples/SharedMemory/PhysicsClientC_API.h

How can I use this? I'm currently working on a Rust binding of Bullet (Rust bindings only really work with C), but it's unclear how to actually use Bullet from C as there are a lot of different physics clients in the examples folder and no clear, simple examples, or guides.

If somebody could give me some guidance on this that'd be awesome.
Thanks

erwincoumans, 2020-02-29:
There are two examples:
Python bindings here:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/pybullet.c

C++ wrapper around this C API:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/b3RobotSimulatorClientAPI_NoDirect.cpp

closed 2020-03-01 by AlexApps99

AlexApps99, 2020-03-01:
How would I build and link a project using the C API with CMake? Trying to compile the examples folder needs OpenGL among other things and it takes too long, all I need is the SharedMemory folder

erwincoumans, 2020-03-01:
premake and cmake have a BulletRobotics project in the Extras folder. It doesn't require OpenGL.

AlexApps99, 2020-03-01:
I keep getting linkage errors while trying to disable parts of Bullet that are unneeded, what do I need to compile BulletRobotics?
I currently have enabled:
- `BUILD_BULLET3`
- `BUILD_EXTRAS`
Every other `BUILD_*` is disabled, I am getting the error `/usr/bin/ld: cannot find -lBussIK`.

What else do I need to enable to get this to work?

AlexApps99, 2020-03-04:
@erwincoumans please could you help me? Thanks :)