bulletphysics/bullet3 #2650: PyBullet: only one color is handled in a body, even with multiple <visual> tags

Gregwar, 2020-03-02:
In the following URDF example, the two boxes appear red, instead of one red and one green when rendered using PyBullet:

```xml
<?xml version="1.0" ?>
<robot name="example" xmlns:xacro="http://www.ros.org/wiki/xacro">
    <link name="main">
        <inertial>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <mass value="1"/>
            <inertia ixx="4E-06" ixy="0" ixz="0" iyy="4E-06" iyz="0" izz="4E-06"/>
        </inertial>
        <visual>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry><box size="0.3 0.3 0.3" /></geometry>
            <material name="green"/>
        </visual>
        <visual>
            <origin rpy="0 0 0" xyz="1 0 0"/>
            <geometry><box size="0.3 0.3 0.3" /></geometry>
            <material name="red"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="1 0 0"/>
            <geometry><box size="0.3 0.3 0.3" /></geometry>
        </collision>
    </link>

    <material name="green"><color rgba="0 1 0 1"/></material>
    <material name="red"><color rgba="1 0 0 1"/></material>
</robot>
```

erwincoumans, 2020-03-02:
Correct, we only support one material per link. The best solution is to bake the colors into a texture instead. Alternatively (slower) is to use fixed joints to split the link per material.

closed 2020-03-02 by erwincoumans

Gregwar, 2020-03-02:
Actually, you do parse the colors, because the following are correct, only it is not displayed in the viewer

```
p.getVisualShapeData(0)[0][7]
p.getVisualShapeData(0)[1][7]
```

erwincoumans, 2020-03-02:
Yes, the TinyRenderer actually renders them green and red as well using getCameraImage.
Is it primarily the visualizer that doesn't support it (and the EGL/hardware render modes of getCameraImage, which use that code path)