bulletphysics/bullet3 #2654: PyBullet: Big STLs are not totally loaded

Gregwar, 2020-03-04:
If an URDF contains too much facets in its STLs, it is only partially loaded
Again, this issue can be closed with some leads on how to fix this in the source

## Example of failing model
[image]
URDF & STLs:
https://gregwar.com/sigmaban/sigmabal_big.tgz

##  The exact same model with meshes simplification works
[image]
URDF & STLs:
https://gregwar.com/sigmaban/sigmabal_light.tgz

erwincoumans, 2020-03-15:
You need to reduce the number of vertices first, there are hardcoded limits.

closed 2020-03-15 by erwincoumans

Nuofan-SUSTech, 2024-06-21:
> You need to reduce the number of vertices first, there are hardcoded limits.

Is there any way to improve the hardcoded limits?