bulletphysics/bullet3 #2658: Improve constraint documentation

jmakitalo, 2020-03-06:
I'm having very hard time understanding how to use constraints based on the documentation and code samples. Basically I understand that a constraint is between two rigid bodies or a rigid body and world but the other parameters are not clear. For example The 6DOF constraint, what exactly are frameInA and frameInB? And in which space are they defined? To which basis vectors of those frames do the limits apply to? And what does useLinearReferenceFrameA mean? I have read the header file and user manual and tried to read the examples but there is no proper explanation of these parameters.

What I think would be best was if there was a graphical illustration where the frames were visualized and how useLinearReferenceFrameA affects the result. Also some illustration how the limits are applied. Even for the hinge constraint you can give a frame but it is not explained how this defines the hinge axis. It's probably one of the basis vectors of the frame but which one?

jmakitalo, 2020-03-09:
It also seems that the user manual is only versioned as a PDF. Would it be possible to have the source as latex or markdown so that it could be more easily developed by community?

erwincoumans, 2020-03-15:
Please use the forums instead: https://pybullet.org

closed 2020-03-15 by erwincoumans

jmakitalo, 2020-03-26:
I don't think the forums is a substitute for documentation.