bulletphysics/bullet3 #2663: Can't control joints

Jingjinganhao, 2020-03-11:
Hi， my robotic arm has 4 degrees of freedom, but it has 8 joint. When  I let it move, some joints broke apart。How can I fix it? Is there some API can make them move as other moving joints。
my urdf is as follow：
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="arm" >

<link name="Base">
</link>
<link name="Link1">
</link>
<link name="Link2">
</link>
<link name="Link3">
</link>
<link name="Link4">
</link>
<link name="Link5">
</link>
<link name="Link6">
</link>
<link name="Link7">
</link>
<link name="Link8">
</link>
<link name="Link9">
</link>
<joint name="Joint1" type="revolute">
    <parent link="Base"/>
    <child link="Link1"/>
</joint>
<joint name="Joint2" type="revolute">
    <parent link="Link1"/>
    <child link="Link2"/>
</joint>
<joint name="Joint3" type="revolute">	
	<parent link="Link2" />
	<child link="Link3" />
</joint>
<joint name="Joint4" type="revolute">
    <parent link="Link2"/>
    <child link="Link4"/>
</joint>
<joint name="Joint5" type="revolute">
    <parent link="Link1"/>
    <child link="Link5"/>
</joint>
<joint name="Joint6" type="revolute">
    <parent link="Link1"/>
    <child link="Link6"/>
</joint>
<joint name="Joint7" type="revolute">
    <parent link="Link6"/>
    <child link="Link7"/>
</joint>
<joint name="Joint8" type="revolute">
    <parent link="Link3"/>
    <child link="Link8"/>
</joint>
<joint name="Joint9" type="revolute">
    <parent link="Link8"/>
    <child link="Link9"/>
</joint>

</robot>
Thanks！

Jingjinganhao, 2020-03-11:
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="arm" >

<link name="Base">
</link>
<link name="Link1">
</link>
<link name="Link2">
</link>
<link name="Link3">
</link>
<link name="Link4">
</link>
<link name="Link5">
</link>
<link name="Link6">
</link>
<link name="Link7">
</link>
<link name="Link8">
</link>
<link name="Link9">
</link>
<joint name="Joint1" type="revolute">
    <parent link="Base"/>
    <child link="Link1"/>
</joint>
<joint name="Joint2" type="revolute">
    <parent link="Link1"/>
    <child link="Link2"/>
</joint>
<joint name="Joint3" type="revolute">	
	<parent link="Link2" />
	<child link="Link3" />
</joint>
<joint name="Joint4" type="revolute">
    <parent link="Link2"/>
    <child link="Link4"/>
</joint>
<joint name="Joint5" type="revolute">
    <parent link="Link1"/>
    <child link="Link5"/>
</joint>
<joint name="Joint6" type="revolute">
    <parent link="Link1"/>
    <child link="Link6"/>
</joint>
<joint name="Joint7" type="revolute">
    <parent link="Link6"/>
    <child link="Link7"/>
</joint>
<joint name="Joint8" type="revolute">
    <parent link="Link3"/>
    <child link="Link8"/>
</joint>
<joint name="Joint9" type="revolute">
    <parent link="Link8"/>
    <child link="Link9"/>
</joint>

</robot>

erwincoumans, 2020-04-01:
URDF doesn't support closed looks, so you need to add constraints to connect parts.
This is a bit more complicated. See the dobot.py example how to do this:
https://github.com/bulletphysics/pybullet_robots/blob/master/dobot.py

closed 2020-04-01 by erwincoumans

Jingjinganhao, 2020-04-07:
Thank you very much 
I did what you said but it didn't work 
I did as follow:
I want to build joint1 and jointtype is revolute
c = p.createConstraint(uarmId,3,uarmId,2,jointType=p.JOINT_POINT2POINT,jointAxis =[0,1,0],
                       parentFramePosition=[0.0274502,0.0165272,0.02698],
                       childFramePosition=[-0.1587598,0.0165272,-0.00006])
parentFramePosition = joint1-joint3
childFramePosition = joint1-joint2
Is this wrong?
What does p.JOINT_POINT2POINT mean? it means revolute?