bulletphysics/bullet3 #2665: The units of nearVal and farVal parameters for computation of Depth Images

gktg1514, 2020-03-12:
We aim to generate RGB and Depth images of objects by using pybullet. Then, we plan to generate 3D point clouds of those objects by using these two kinds of images. Unfortunately, there is no built-in function for point cloud generation in pybullet. 

However, we do not know the measure units of nearVal and farVal parameters used in computation of projection matrix. Is it millimeter or meter ?

erwincoumans, 2020-03-12:
See the example here with point clouds, it has some info about near/far:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py

For discussion, please use the forums at http://pybullet.org

closed 2020-03-12 by erwincoumans

bestgodok, 2021-03-16:
> We aim to generate RGB and Depth images of objects by using pybullet. Then, we plan to generate 3D point clouds of those objects by using these two kinds of images. Unfortunately, there is no [...]

Do you know how to generate the Depth images?I used the API getCameraImage, but  it returns values between 0 and 1

jamesbraza, 2021-12-25:
@GoktugGuvercin did you ever figure out the units?  I have the same question as you.

Looking through the `pointCloudFromCameraImage.py` file there is not a single mention of units, and I am not finding any relevant forum posts either.

jamesbraza, 2022-01-02:
Okay, as suggested above, I made a forum post here: https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=9&t=13345

---

Also, to answer @bestgodok:

> I used the API getCameraImage, but it returns values between 0 and 1

`getCameraImage` returns a 5-item `tuple` object, I am not sure what you did.  Try downloading and locally running the [`pointCloudFromCameraImage.py`](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py) or [`getCameraImageTest.py`](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/getCameraImageTest.py) files with a debugger, and then inspect the return from `getCameraImage`... should return a bunch of things.  Personally, I find PyCharm's built-in debugger is great.

Also, for what it's worth, `getCameraImage`'s returns are documented in the [PyBullet Quickstart Guide here](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.u1jisfnt6984).  Hope this helps.