bulletphysics/bullet3 #2666: Can't install pybullet on Windows 10

aadevmobile, 2020-03-12:
Hello,
I'm trying to install qiBullet but it keeps failing when to installing pybullet.

This is what I get:
[image]
[image]

Did you see this before? Any suggestions? I'm struggling with this..

I've read 1708. Mine doesnt say anything about stdint.h, i've tried it tho, no luck.

KevinJune, 2020-03-13:
[image]
[image]
[image]

KevinJune, 2020-03-13:
I have similar problem with you, and I solve it by deleteing the vs2015 completely(both softwares about vs and the environment variable)
Then I only install the Build tool box which erwin said.
[https://download.microsoft.com/download/5/f/7/5f7acaeb-8363-451f-9425-68a90f98b238/visualcppbuildtools_full.exe?fixForIE=.exe.](url)
Finally you can use a program to test whether you install the tool box correctly.
Hope my answer can help you.

erwincoumans, 2020-03-15:
Please use the forums instead: https://pybullet.org

closed 2020-03-15 by erwincoumans

aadevmobile, 2020-03-17:
Hi Erwin,

https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=9&t=12890
Could you please help me? I've tried so many things, it's not working...

erwincoumans, 2020-03-17:
I fixed the issues, can you please try again with pybullet-2.6.9 and report back.
Also uploaded a precompiled wheel for Python 2.7 Windows 64bit.

Thanks!
Erwin

aadevmobile, 2020-03-17:
Thank you! Thank you! Thank you!
It works now!!

aadevmobile, 2020-03-18:
I was too happy too fast...
I need to have it working for Python 2.7 Windows 32bits now :/
I've updated the info in the forum... It seems to be a different error now.
Or it's something I'm doing very wrong and I don't know how to do right...

Could you please take a look at again?