bulletphysics/bullet3 #2668: Ccd introduces too high impulses too easily in 2.89, leads to floating point errors. Patch draft included [needs proper rework]

etc0de, 2020-03-13:
I have played around with bullet 2.89 in a custom game engine, and hit a brick wall with Ccd:

Enabling Ccd/anti tunneling in some simple but non-trivial setups (like, only boxes in somewhat reasonable sizes with all sides significantly larger than the collision margin, and triangle meshes as static level geometry but the geometry not being super trivial) can occasionally introduce too high impulses too easily in bullet 2.89, leading to floating point errors. Occasionally sadly means it *will* happen at some point if there's just enough activity, and objects fall down from high enough often enough in a large enough pile.

When these floating point errors happen, it leads to a `NaN` injection which poisons all the positions and/or velocities, and propagates brokenness into all objects:

:arrow_right: [Video illustrating the issue](https://wobble.ninja/temp/test_windows_omgwat.mp4)

I found this to be almost unavoidable in any non-trivial scenario. With only using really large objects (>1.0 only) it is *somewhat* avoidable, but games often have smaller inventory items so I find not using anything at least down to size 0.3 x 0.3 x 0.3 extremely limiting in practice. Disabling Ccd worked also, but that obviously leads to lots of tunneling again, especially for these smaller objects. Often having a pile of them fall down onto a triangle mesh ground is enough for half of them to pass through!

This sadly makes Ccd way less useful, because in less trivial scenarios tunneling can be pretty bad but these scenarios also appear to like to cause `NaN` breakages with Ccd enabled. :cry: 

Therefore, I enabled floating point exceptions and devised this patch to more permanently address this:

```
diff --git a/src/BulletDynamics/Dynamics/btRigidBody.cpp b/src/BulletDynamics/Dynamics/btRigidBody.cpp
index 9e8705b00..684b570f3 100644
--- a/src/BulletDynamics/Dynamics/btRigidBody.cpp
+++ b/src/BulletDynamics/Dynamics/btRigidBody.cpp
@@ -294,6 +294,7 @@ btVector3 btRigidBody::computeGyroscopicImpulseImplicit_Body(btScalar step) cons
 {
 	btVector3 idl = getLocalInertia();
 	btVector3 omega1 = getAngularVelocity();
+	limitForceOrVel(omega1);
 	btQuaternion q = getWorldTransform().getRotation();
 
 	// Convert to body coordinates
diff --git a/src/BulletDynamics/Dynamics/btRigidBody.h b/src/BulletDynamics/Dynamics/btRigidBody.h
index 39d47cbbd..0d103af61 100644
--- a/src/BulletDynamics/Dynamics/btRigidBody.h
+++ b/src/BulletDynamics/Dynamics/btRigidBody.h
@@ -305,6 +305,7 @@ public:
 	void applyTorque(const btVector3& torque)
 	{
 		m_totalTorque += torque * m_angularFactor;
+		limitForceOrVel(m_totalTorque);
 	}
 
 	void applyForce(const btVector3& force, const btVector3& rel_pos)
@@ -316,11 +317,13 @@ public:
 	void applyCentralImpulse(const btVector3& impulse)
 	{
 		m_linearVelocity += impulse * m_linearFactor * m_inverseMass;
+		limitForceOrVel(m_linearVelocity);
 	}
 
 	void applyTorqueImpulse(const btVector3& torque)
 	{
 		m_angularVelocity += m_invInertiaTensorWorld * torque * m_angularFactor;
+		limitForceOrVel(m_angularVelocity);
 	}
 
 	void applyImpulse(const btVector3& impulse, const btVector3& rel_pos)
@@ -362,19 +365,34 @@ public:
         m_pushVelocity = v;
     }
     
+    void limitForceOrVel(btVector3& v) const {
+        v.setX(
+            fmax(-9999, fmin(9999, v.getX()))
+        );
+        v.setY(
+            fmax(-9999, fmin(9999, v.getY()))
+        );
+        v.setZ(
+            fmax(-9999, fmin(9999, v.getZ()))
+        );
+    }
+
     void setTurnVelocity(const btVector3& v)
     {
         m_turnVelocity = v;
+        limitForceOrVel(m_turnVelocity);
     }
     
     void applyCentralPushImpulse(const btVector3& impulse)
     {
         m_pushVelocity += impulse * m_linearFactor * m_inverseMass;
+        limitForceOrVel(m_pushVelocity);
     }
     
     void applyTorqueTurnImpulse(const btVector3& torque)
     {
         m_turnVelocity += m_invInertiaTensorWorld * torque * m_angularFactor;
+        limitForceOrVel(m_turnVelocity);
     }
 
 	void clearForces()
@@ -408,12 +426,14 @@ public:
 	{
 		m_updateRevision++;
 		m_linearVelocity = lin_vel;
+		limitForceOrVel(m_linearVelocity);
 	}
 
 	inline void setAngularVelocity(const btVector3& ang_vel)
 	{
 		m_updateRevision++;
 		m_angularVelocity = ang_vel;
+		limitForceOrVel(m_angularVelocity);
 	}
 
 	btVector3 getVelocityInLocalPoint(const btVector3& rel_pos) const
```
This works surprisingly well. It won't prevent instabilities / wildly exploding around items, but it prevents the floating point exceptions and permanent `NaN` breakage in my tests even when things briefly go wild.

I recognize this patch can't go in like this, because it has performance implications and changes the simulation outcome in ways some people might not like. But I think this as an option would help really greatly in making Ccd are more practically useful option, since right now I find it hard to find much use in it given these numeric issues.

Maybe this could be added to premake.lua with a sane option, like `--hard-clamp-all-velocity`? (by someone who knows how to add such options)

etc0de, 2020-03-13:
I just the drafted changes as a pull request in this fork if anyone wants to look at it in GitHub UI: https://github.com/etc0de/bullet3/pull/1

(I didn't create a pull request here, since I assume it's not in a shape useful anyway. I invite anyone to copy it and use it as a base for whatever sort of pull request that is reasonable)

erwincoumans, 2020-03-13:
Thanks, this may help others.
Sure. you may want to turn the 9999 into some variable that can be set (dispatch Info or so).

etc0de, 2020-03-13:
Ok, I now made a more proper pull request with it: https://github.com/bulletphysics/bullet3/pull/2669

As for dispatch info, is that easily available from the btRigidBody (which may not be attached to a world)? In overall I think most would want this to be a compile time option anyway, due to the performance implications since this runs additional code every time velocity is changed which I assume adds significant slowdowns. So I made it a compile time option for now

commit 6985f06e (2020-04-26): Merge pull request #2669 from etc0de/velocity-limits

Add a compile time option to clamp velocity to reduce floating point exception risk, fixes #2668
closed 2020-04-26 by erwincoumans