bulletphysics/bullet3 #2675: Strange behavior of btSoftRigidDynamicsWorld::rayTest()

consultit, 2020-03-16:
I'm used to building bullet3 directly from git source and I found this issue on  the last few commits.
My code performs closest "rayTest" for various purposes:
```
...
btCollisionWorld::ClosestRayResultCallback rayCallback (rayFromWorld, rayToWorld);
softDynamicsWorld->rayTest (rayFromWorld, rayToWorld, rayCallback);
...
body = (btRigidBody*) btRigidBody::upcast(rayCallback.m_collisionObject);
sBody = (btSoftBody*) btSoftBody::upcast(rayCallback.m_collisionObject);
...
```
with `body` and `sBody` that check the kind of hit object.
Now, when a soft body is closer than a rigid body, the very first time the call to `rayTest()` reports (uncorrectly) that `body != nullptr` and` sBody == nullptr`, while all subsequent calls report (correctly) that `body == nullptr` and `sBody != nullptr`.
Before this did not happen.

erwincoumans, 2020-03-16:
I accepted some large PRs and it must have slipped in. Can you help tracking down what commit and what file(s) need to be reverted?

xhan0619, 2020-03-16:
Could you provide a code snippet to reproduce the bug? I can help look into it. Thanks.

consultit, 2020-03-17:
Using `git bisect` and after a series of code builds, this is the result:
```
3bc701c487bf558ec68c48f308848a80df3d431f is the first bad commit
commit 3bc701c487bf558ec68c48f308848a80df3d431f
Author: Xuchen Han <xuchenhan2015@u.northwestern.edu>
Date:   Thu Feb 6 16:58:35 2020 -0800

    Improve intra-deformable collision and deformable self collision

:040000 040000 a6bcf0357dafaa5c3d94f073fd6aa87d28061d3d eaa2bc90322921832067a8387c192e63acee70d2 M      examples
:040000 040000 b3d3e1fabc68ccc8f31e620593c2eba905f57593 d92540ab9c942ce77ca096168af17c8457845767 M      src
```
ie this commit should have introduced the issue.

erwincoumans, 2020-03-17:
Thanks a lot for tracking down the commit. Do you have a small reproduction case?

consultit, 2020-03-18:
To reproduce the problem I took the following steps:
- I pulled down the last commit of the `master` branch (`3c9dde54`).
- I applied this [issue-patch.txt](https://github.com/bulletphysics/bullet3/files/4348279/issue-patch.txt) patch using `git apply issue.patch` from the source root
- I built from scratch using `premake64`

At this point it was enough to run `App_BulletExampleBrowser` and do some tests by clicking on the various soft bodies to highlight the issue.
I noticed that it only seems to work well with "tetra" bodies.

xhan0619, 2020-03-20:
@consultit, thanks for the repro code. I'll look into it.

erwincoumans, 2020-03-22:
@consultit can you verify this fixes it?
https://github.com/bulletphysics/bullet3/pull/2685

consultit, 2020-03-22:
I have tested and everything works again. Well done

closed 2020-03-22 by consultit