bulletphysics/bullet3 #2684: Question about the parameters of btMultiBody::fillConstraintJacobianMultiDof()?

Steven89Liu, 2020-03-20:
i'm confused about this part of code as below. do the normal_lin_world and normal_ang_world can be the zero at the same time? thanks.
``` 
       btVector3 omega_coeffs_world;
	omega_coeffs_world = p_minus_com_world.cross(normal_lin_world);
	jac[0] = omega_coeffs_world[0] + normal_ang_world[0];
	jac[1] = omega_coeffs_world[1] + normal_ang_world[1];
	jac[2] = omega_coeffs_world[2] + normal_ang_world[2];
```

erwincoumans, 2020-03-26:
if they are both zero it would be a useless jacobian

closed 2020-03-26 by erwincoumans

Steven89Liu, 2020-03-27:
sorry, i made a mistake, i was thinking to ask if they can be non-zero at the same time. thanks.

Steven89Liu, 2020-04-04:
update. hope you can see my last comment that is important for me. thanks.