bulletphysics/bullet3 #2693: Bad Inertia Tensor Properties

Berk035, 2020-03-26:
Hello dear contributors;

I deal with an error while I am debugging my code. It appears at **only debug** mode.The error:

`b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,358]:
Bad inertia tensor properties, setting inertia to zero for link: base_link` 

My robot can not introduced regularly due to this problem and I can't move on the study. Do you have any idea for solution?

Thank you in advance. Have a healthy days.

erwincoumans, 2020-03-26:
The URDF file must have bad or invalid inertia properties. Share the URDF and someone may be able to help. Just check the <inertia> fields of existing URDF files, there are plenty in the Bullet repository:
https://github.com/bulletphysics/bullet3/tree/master/data
and
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/gym/pybullet_data

closed 2020-03-26 by erwincoumans

Berk035, 2020-03-27:
Thanks for your response. It doesn't match with bullet's models in my opinion.
Have a nice days.