bulletphysics/bullet3 #2694: Segfault

richardrl, 2020-03-29:
```
    def get_pos(self):
        """
        Return the joint position(s) of the gripper.

        Returns:
            float: joint position.
        """
        with self.lock:
            pos = self._pb.getJointState(self.robot_body_id, jnt_id)[0]
        return pos
```

I am running the above function in a main thread and in a child thread (Threading.Thread).

`            pos = self._pb.getJointState(self.robot_body_id, jnt_id)[0]
`

I am getting a segfault:
```
(airobot_venv) Richards-MacBook-Pro-7:airobot richard$ python examples/ur5e/sim/block_stacking.py 
pybullet build time: Mar 28 2020 16:47:39
Version = 4.1 NVIDIA-10.17.5 355.10.05.45f01
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce GT 750M OpenGL Engine
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
Client ID0
Client ID0
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: ee_link
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: base
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: tool0
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: left_inner_finger_pad
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: right_inner_finger_pad
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: ee_tip
Get_pos
Get_pos
Fatal Python error: Segmentation fault

Current thread 0x00007000043f3000 (most recent call first):
  File "/Users/richard/improbable/airobot/src/airobot/ee_tool/robotiq2f140_pybullet.py", line 165 in get_pos
  File "/Users/richard/improbable/airobot/src/airobot/ee_tool/robotiq2f140_pybullet.py", line 223 in _set_rest_joints
  File "/Users/richard/improbable/airobot/src/airobot/ee_tool/robotiq2f140_pybullet.py", line 214 in _th_mimic_gripper
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763 in run
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810 in __bootstrap_inner
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 783 in __bootstrap

Thread 0x00007fffb2ca33c0 (most recent call first):
  File "/Users/richard/improbable/airobot/src/airobot/__init__.py", line 147 in __init__
  File "examples/ur5e/sim/block_stacking.py", line 15 in main
  File "examples/ur5e/sim/block_stacking.py", line 58 in <module>
Segmentation fault: 11

```

I have narrowed down the error to the getJointState calls. If I disable the child thread, things work fine in the main thread (of course I am losing certain functionality I put in the child thread).

erwincoumans, 2020-03-29:
One pybullet client (or bullet_client.py) should only be called from 1 thread only. You can make calls from 1 more different thread of process by using a new pybullet instance that connects to the server using SHARED_MEMORY.

richardrl, 2020-04-02:
Hi Erwin,
Thanks for your response. 

I am currently getting -1 when trying to connect with pybullet.SHARED_MEMORY in my child thread. Is there any way to get a verbose debug message why?

richardrl, 2020-04-02:
Here is a minimum example showing the error. Running this errors:

```
Traceback (most recent call last):
  File "fingertip_test.py", line 40, in <module>
    joint_names_to_idx(pgui)['left_inner_finger_joint'])
pybullet.error: GetJointInfo failed.
```

[tests.zip](https://github.com/bulletphysics/bullet3/files/4423709/tests.zip)

erwincoumans, 2020-04-02:
Don't have bandwidth to support those issues, please use the forums instead.

closed 2020-04-02 by erwincoumans