bulletphysics/bullet3 #2697: SharedMemory_GUI + SharedMemoryUDPBridge with Unity3D and Ubuntu clients

missxa, 2020-03-31:
I'm trying to run pybullet crossplatform with the following setup:
Windows machine:
- SharedMemory_GUI
- SharedMemoryUDPBridge
- Unity3D client in shared memory mode (here I'm using autogenerated `NativeMethods.cs` from unity3d example)

Ubuntu machine:
- python client in UDP mode

`SharedMemory_GUI` and `SharedMemoryUDPBridge` start as expected.
When I start the Unity3D app, it successfully connects to pybullet server, but then when I try to connect from Ubuntu python client, Windows' `SharedMemory_GUI` segfaults on `b3ResizablePool.h:77`.

Can this setup work theoretically? Any pointers will be helpful, thanks!

erwincoumans, 2020-03-31:
It should work, but UDP isn't tested well. I would first try TCP instead.
Also, make sure the builds are both using 64bit and double precision.

missxa, 2020-04-01:
@erwincoumans thanks for the hint to look at TCP, it indeed connected, but I ran into another issue. 

To simplify the testing, I'm running the following setup (Windows machine only, also no Unity stuff for now):
- `PhysicsServer_SharedMemory_GUI`
- `PhysicsServerSharedMemoryBridgeTCP`
- python client1 with `p.connect(p.SHARED_MEMORY)`
- python client2 with `p.connect(p.TCP)`

When I call `loadURDF` from client1, it works as expected, but when then from client2 I call `getNumBodies`, it always returns 0 (for client1 it's 1). Why could that be? Shouldn't they be in sync?

erwincoumans, 2020-04-01:
This is expected. Can you call pybullet.syncBodyInfo and try to getNumBodies again?

erwincoumans, 2020-04-01:
Note that you can only have 1 client connect to each shared memory section. There are 2 shared memory sections. The TCP bridge uses shared memory key1.
So you need to connect using the second key:
p.connect(p.SHARED_MEMORY, p.SHARED_MEMORY_KEY2)
#define SHARED_MEMORY_KEY 12347
SHARED_MEMORY_KEY2 = SHARED_MEMORY_KEY+1
Note you can use pybullet as server, using pip install pybullet, then python -m pybullet_utils.runServer

missxa, 2020-04-01:
> This is expected. Can you call pybullet.syncBodyInfo and try to getNumBodies again?

when I loadURDF from the TCP client and in the shared memory client call syncBodyInfo and getNumBodies - it all works, but vice versa - the server crashes.

missxa, 2020-04-01:
> Note that you can only have 1 client connect to each shared memory section. There are 2 shared memory sections. The TCP bridge uses shared memory key1.
> So you need to connect using the second key: [...]

ok, this is helpful, will take it into account

erwincoumans, 2020-04-01:
Should be fine then.

closed 2020-04-01 by erwincoumans

missxa, 2020-04-02:
yup, works, thanks!