bulletphysics/bullet3 #2698: The conservation of momentum is broken in btMultibody

Rullec, 2020-03-31:
Hi all,

   I am using bullet as a physics enignee in my application.But when I try to verify the conservation law of (linear) momentum for btMultibody structure, the result is not in my expection.
## Question Description
   Briefly, I create a btMultibody with 3 links, disable all friction coeff, simulating it along with adding joint torque on the first spherical joint, then compare the linear momentum changes with external gravity impulse.In my opinion, The linear momentum of this btMultibody should reply on nothing except the gravity, isn't it?(If I am wrong please point it out).

  I use The examples/MultiBody/MultiDofDemo.cpp for making a minimum reproduce code. If you guys **paste all the following code stuffs in MultiDofDemo.cpp** then run it, maybe you can get the same output as me...I dont know..

  P.S. This project is based on macOS Mojave. 

  ANY suggestion is welcome!
## The experiment result 
```
b3Printf: old momentum -17.42704, -0.55479, 6.60006
b3Printf: new momentum -20.20674, -0.38032, 7.95865
b3Printf: gravity impulse 0.00000, -0.03000, 0.00000
b3Printf: momentum changes -2.77970, 0.17447, 1.35859
b3Printf: relative error = 10335.655%
```
**We can see that the momentum changes is significantly different from the impulse.**

## minium reproduce code
```
#include "MultiDofDemo.h"
#include "../OpenGLWindow/SimpleOpenGL3App.h"
#include "btBulletDynamicsCommon.h"

#include "BulletDynamics/MLCPSolvers/btDantzigSolver.h"
#include "BulletDynamics/MLCPSolvers/btLemkeSolver.h"
#include "BulletDynamics/MLCPSolvers/btSolveProjectedGaussSeidel.h"

#include "BulletDynamics/Featherstone/btMultiBody.h"
#include "BulletDynamics/Featherstone/btMultiBodyConstraintSolver.h"
#include "BulletDynamics/Featherstone/btMultiBodyMLCPConstraintSolver.h"
#include "BulletDynamics/Featherstone/btMultiBodyDynamicsWorld.h"
#include "BulletDynamics/Featherstone/btMultiBodyLinkCollider.h"
#include "BulletDynamics/Featherstone/btMultiBodyLink.h"
#include "BulletDynamics/Featherstone/btMultiBodyJointLimitConstraint.h"
#include "BulletDynamics/Featherstone/btMultiBodyJointMotor.h"
#include "BulletDynamics/Featherstone/btMultiBodyPoint2Point.h"
#include "BulletDynamics/Featherstone/btMultiBodyFixedConstraint.h"
#include "BulletDynamics/Featherstone/btMultiBodySliderConstraint.h"

#include "../OpenGLWindow/GLInstancingRenderer.h"
#include "BulletCollision/CollisionShapes/btShapeHull.h"

#include "../CommonInterfaces/CommonMultiBodyBase.h"
#include <iostream>

btVector3 gGravity = btVector3(0, -10, 0);
class MultiDofDemo : public CommonMultiBodyBase
{
public:
	MultiDofDemo(GUIHelperInterface* helper);
	virtual ~MultiDofDemo();

	virtual void initPhysics();

	virtual void stepSimulation(float deltaTime);

	virtual void resetCamera()
	{
		float dist = 1;
		float pitch = -35;
		float yaw = 50;
		float targetPos[3] = {-3, 2.8, -2.5};
		m_guiHelper->resetCamera(dist, yaw, pitch, targetPos[0], targetPos[1], targetPos[2]);
	}

	btMultiBody* createFeatherstoneMultiBody_testMultiDof(class btMultiBodyDynamicsWorld* world, int numLinks, const btVector3& basePosition, const btVector3& baseHalfExtents, const btVector3& linkHalfExtents, bool spherical = false, bool floating = false);
[... 457 lines cut ...]
		{
			int multibody_link_id = full_link_id - 1;
			auto & cur_trans = multibody->getLinkCollider(multibody_link_id)->getWorldTransform();
			link_vel_world[full_link_id] = quatRotate(cur_trans.getRotation(), vel_buffer[full_link_id]);
			linear_momentum += link_vel_world[full_link_id] * multibody->getLinkMass(multibody_link_id);
		}
	}

	// 3. return
	return linear_momentum;
}

void MultiDofDemo::addJointTorque()
{
	btVector3 linear_momentum = btVector3(0, 0, 0);
	auto multibody = m_dynamicsWorld->getMultiBody(0);
	btScalar torque[3] = {10, 10, 10};
	multibody->addJointTorqueMultiDof(0, torque);
}

class CommonExampleInterface* MultiDofCreateFunc(struct CommonExampleOptions& options)
{
	return new MultiDofDemo(options.m_guiHelper);
}
```

erwincoumans, 2020-04-01:
Please use the forums for discussions. Also, better to use PyBullet in Python, so your repro case can be 10 lines instead.

closed 2020-04-01 by erwincoumans

Rullec, 2020-04-03:
> Please use the forums for discussions. Also, better to use PyBullet in Python, so your repro case can be 10 lines instead.

I will migrate this problem to the forum, thanks a lot! And the pybullet repro case is on the way.