bulletphysics/bullet3 #2702: ModuleNotFoundErro: No module named 'tf.compat'; 'tf' is not a package

The132, 2020-04-01:
after the code update "Replace tensorflow with tf.compat.v1"，I meet a problem when run "python3 minitaur_trotting_env_example.py",The terminal presents the following error "ModuleNotFoundErro: No module named 'tf.compat'; 'tf' is not a package". how can I solve it.

erwincoumans, 2020-04-01:
It was a commit that doesn't support backwards compatibility. We need to add try/except statements instead. TensorFlow keeps on breaking the API.
~~~~~~~~~~~~~~~~~~~
try:
  import tensorflow.compat.v1 as tf
except Exception:
  import tensorflow as tf
~~~~~~~~~~~~~~~~~~~

The132, 2020-04-03:
Thanks for your answer and hope this work can be completed as soon as possible，thank you.

erwincoumans, 2020-04-06:
Sounds good, we'll await your contribution.

closed 2021-04-26 by erwincoumans