bulletphysics/bullet3 #2703: compile issue thgirdparty clsocket linux.

stolk, 2020-04-01:
```
In file included from /home/bram/src/bullet3/examples/ThirdPartyLibs/clsocket/src/SimpleSocket.h:85:0,
                 from /home/bram/src/bullet3/examples/ThirdPartyLibs/clsocket/src/ActiveSocket.h:46,
                 from /home/bram/src/bullet3/examples/SharedMemory/RemoteGUIHelperTCP.cpp:17:
/home/bram/src/bullet3/examples/ThirdPartyLibs/clsocket/src/StatTimer.h:90:2: error: ‘uint32’ does not name a type; did you mean ‘uint64’?
  uint32 GetMilliSeconds() { return (CalcTotalUSec() / MILLISECONDS_CONVERSION); };
  ^~~~~~
  uint64
```

Config:
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Linux
CMake Warning (dev) at /home/bram/share/cmake-3.15/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:287 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
OPENGL FOUND
/usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bram/src/bullet3/build
```

erwincoumansepic, 2020-04-01:
Thanks Bram,
I've not seen the clsocket thing when building on Linux. Did you use cmake without any further flags? What compiler version?

We could remove FindOpenGL, since we don't really need it (we use glad, which dynamically loads OpenGL and has its own headers).

stolk, 2020-04-01:
Erwin,
I apologize! I messed up. Years ago, I created a cmake config that disabled clsocket and enet, which used to work. If I remove those defines, it works again.
```
cmake \
        -DBULLET2_MULTITHREADING=ON \
        -DBULLET2_USE_TBB_MULTITHREADING=ON \
        -DBUILD_CLSOCKET=OFF \
        -DBUILD_ENET=OFF \
        -DCMAKE_BUILD_TYPE=Release ../
```

erwincoumans, 2020-04-01:
Well, it it true that some flags combinations are not working indeed. (myself, I've mostly used premake and the Python setuptools (setup.py) and pip last few years). There is some cmake refactor pull request, but there are so many changes in there that I don't even took the time to review it). 

Can we close this?

closed 2020-04-01 by stolk