bulletphysics/bullet3 #2705: 6DoF Spring constraint changes.

stolk, 2020-04-01:
It's been 2 yrs or so, since I updated my bullet copy, which caused my sim to break down.

It now looks like this:

https://www.youtube.com/watch?v=O1DZOjExAL0&feature=youtu.be

Whereas with the old code, it was working just fine:

https://twitter.com/BramStolk/status/1245207361075253249?s=20

Could this have been caused by this change?

https://github.com/bulletphysics/bullet3/commit/cec71341adcee78503d64886a44842a7e953dc19

stolk, 2020-04-01:
Sorry, I had the wrong commit link originally.

stolk, 2020-04-01:
So yeah, after reverting `btGeneric6DofSpring2Constraint.cpp` and `btGeneric6DofSpring2Constraint.h` to the old versions of 2 yrs ago, it works again, so infinite error can cause some serious stability issues.

erwincoumans, 2020-04-01:
Thanks for the update. Could you help by re-introducing some flag that toggles between the old (for you working version) and the current version?

By the way, I think such tank would work nice with btMultiBody, but that would require much more change. If I ever find time to prototype such tank in PyBullet I'll reply here.

stolk, 2020-04-02:
I put the flag back, before I create another pull rq, there is still a pull rq from this morning pending, maybe keep them separate, and I create a new rq tomorrow?

stolk, 2020-04-02:
never mind, github already merged those changes into a single rq. I guess there is always just 1 pull rq active.

erwincoumans, 2020-04-21:
Thanks, I suppose it is fixed.

closed 2020-04-21 by erwincoumans