bulletphysics/bullet3 #2717: How to attach two objects together in PyBullet?

sahandrez, 2020-04-06:
Hello, 

I would like to know how can I attach a shape (e.g. loaded from obj) to a URDF object in PyBullet? I am aware of `createMultiBody` method. But it seems it does not work with an already loaded URDF object. I don't want to modify the URDF directly, because there are many objects being attached to the URDF and I would like to do it inside PyBullet. 

Thank you,
Sahand

erwincoumans, 2020-04-06:
The best way is to use the pybullet_utils.urdfEditor as in this example. It will use the createMultiBody command with all the children etc and allow to join two urdfs in-memory.
See https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/examples/combineUrdf.py

Another option is using createUserConstraint, it will use the constraint solver and hence possible not very rigid (if masses are different, time step large, constraint solver iterations low etc).

closed 2020-04-06 by erwincoumans

sahandrez, 2020-04-06:
Thank you very much for your quick reply!