bulletphysics/bullet3 #2725: The API "btmultibody::getAngularMomentum" is supposed to be wrong

Rullec, 2020-04-08:
Hi all,
 I am reconsidering **the calculation of angular momentum for btMultibody**, which is mainly this function "btMultibody::getAngularMomentum()"
https://github.com/bulletphysics/bullet3/blob/25cc1fa386447771e3a19c1470578a995eb06f9e/src/BulletDynamics/Featherstone/btMultiBody.cpp#L609-L631

  This function shows above means that, the whole angumar momentum is the sum of angular moment of each link with respect to their own link COM individually. 
<img src="https://render.githubusercontent.com/render/math?math=L_{total} = \sum_i R_i I_{body} R_i \omega_i">

But according to this doc http://www.cc.gatech.edu/fac/karenliu/Home_files/angmomentum.pdf (and also my opinion), The total angular momentum should be defined with respect to the COM of this system, which is like:
<img src="https://render.githubusercontent.com/render/math?math=L_{total} = \sum_i \left[ m_i (c_i - c)\times v_i   \space plus \space R_i I_{body} R_i \omega_i \right]">
The strange string "plus" in the formula above is just a normal "+", but I failed to type it in the latex, sorry.

Would you please give me some advice on it? If my derivation is correct, I would be glad to fix it and have a PR.

Thanks a lot!

erwincoumans, 2020-04-10:
I've never used this method, it was inherited from the original multibody code. Happy to just delete it. Please discuss further in the forums.

closed 2020-04-10 by erwincoumans

Rullec, 2020-04-10:
> This is an unused old method that was inherited from the original multibody code.
> Let's remove it: #2736
> Please discuss further in the forums.

So sorry for this disturbution! I really do not want to waste your time :(
Is there any guideline to know which type of questions are suitable in the forum? I hope I can have a correct choice next time.
Thanks you!