bulletphysics/bullet3 #2726: Pybullet crashes on loading meshes for soft body

sahandrez, 2020-04-08:
Hello, 

I would like to load a cube with volumetric mesh as a soft body in Pybullet. I created the mesh with Gmsh and it has a comibnation of points, lines, triangles, and tetrahedras. I export the mesh as .vtk file, but Pybullet crashes on loading it. 

I saw the example meshes in bullet3/data and all of them consist of only tetrahedras. Is that a requirement for Pybullet and the reason for the crash? If so, what could be a workaround? Apparently meshing a cube with only tetrahedras is not that easy (although I don't have much experience in meshing, so there might be some ways to do it).

Thank you!

erwincoumans, 2020-04-08:
Probably only tetrahedral meshes are supported, but we should avoid a crash.
Can you attach a zipped vtk file that crashes?

sahandrez, 2020-04-08:
The attached vtk file causes segmentation error on my end. I'm loading it following how  [deformable_ball.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/deformable_ball.py) loads a soft body.

[cube.zip](https://github.com/bulletphysics/bullet3/files/4453278/cube.zip)

erwincoumans, 2020-04-09:
Only tetrahedra are supported. Xuchen added a check for it to avoid a crash.
https://github.com/bulletphysics/bullet3/pull/2731

closed 2020-04-09 by erwincoumans

HenningLuo, 2020-11-26:
Hi, I am having the same problem generating vtk files with gmsh. The mesh I generated always contain triangles and I have been searching for a few days but haven't had luck. This may be too much to ask, but is it possible if any bullet developers can shed some light on how to solve this? Many thanks.

xhan0619, 2020-11-26:
@HenningLuo, could you briefly outline your workflow for generating the vtk files? I usually hand build the surface mesh and convert it to volume mesh in .msh format with TetWild and then convert to vtk format with Gmsh.

HenningLuo, 2020-11-26:
Hi Xuchen, thanks for your quick response! I have tried both using the GUI of Gmsh and its python package to build the mesh, they reached the same result. My python code is below.
`import gmsh
from math import pi

# Initialise Gmsh
gmsh.initialize()

gmsh.model.add("cylinder")

# Set parameters
gmsh.option.setNumber("Mesh.Algorithm3D", 1)
gmsh.option.setNumber("Mesh.MeshSizeMin", 0.4)
gmsh.option.setNumber("Mesh.MeshSizeMax", 0.4)
gmsh.option.setNumber("Mesh.MeshSizeFactor", 0.1)

# Define a cylinder with opencascade
x, y, z = 0,0,0
dx, dy, dz = 0,0,1
radius = 0.1
gmsh.model.occ.addCylinder(x, y, z, dx, dy, dz, radius, tag = 1, angle = 2*pi)

# Synchronise the data with the Gmsh model:
gmsh.model.occ.synchronize()

# Add physical volumes
gmsh.model.addPhysicalGroup(3, [1], 1)
gmsh.model.setPhysicalName(3, 1, "The volume")

# Generate and save the mesh:
gmsh.model.mesh.generate(3)
# gmsh.model.mesh.removeDuplicateNodes()
gmsh.write("cylinder.vtk")

# Launch the GUI to see the results:
gmsh.fltk.run()
gmsh.finalize()
`
As a matter of fact, I have found an answer from the manual of Gmsh. It says 

> By default, if physical groups are defined, the output mesh only contains those elements
that belong to physical entities. So to save only 3D elements, simply define one (or more)
physical volume(s) and don’t define any physical surfaces, physical curves or physical points.

However, when I add physical volumes in my python codes and in GUI, 2D and 1D elements are still generated. I assume I must have made some mistakes during the process but wasn't able to identify them. 
Any hint is appreciated! I will also try your workflow as you mentioned. Thanks again for your reply and await your insights.

HenningLuo, 2020-11-26:
@xhan0619 Thanks so much for sharing your workflow! Iit works well! The vtk file generated now does not contain 1D and 2D elements anymore! Problem solved!

erwincoumans, 2020-11-26:
Thanks @HenningLuo , what part fixed your issue exactly?

HenningLuo, 2020-11-27:
Hi @erwincoumans. Using tetwild instead of Gmsh to generate the 3D mesh fixed my problem. For anybody else who is just starting to build 3D models, you can generate your surface mesh in .obj or .stl with any software, grow it to volume mesh with tetwild, then convert to .vtk with Gmsh. Lastly, just feed it to your pybullet environment and enjoy!

aprath1, 2021-12-03:
@HainingLuo or @xhan0619 ,
Could you guys kindly share the OS and version for setting up and installing TetWild, can we do it in Ubuntu?
Will be much appreciated. 
So seems like TetWild is the only library which will be supported by Pybullet softbody functionality and should be I guess mentioned as a direct dependency may be in the Documentation. I tried other libs like - VTK, tetgen, gmsh and none of the resultant .msh and .vtk files were not compatible with Pybullet softbody loading (Unless I am missing something here!).

xhan0619, 2021-12-05:
I personally use Ubuntu 18.04 and 20.04. I haven't tried compiling in Windows, but according to their github page windows should be supported (https://github.com/Yixin-Hu/TetWild).

It doesn't matter where the geometry asset comes from as long as the format conforms to the vtk reader in bullet. See, for example, a geometry created from Gmsh for example: https://github.com/bulletphysics/bullet3/blob/5ae9a15ecac7bc7e71f1ec1b544a55135d7d7e32/data/ball.vtk

aprath1, 2021-12-06:
Thank you so much for your response @xhan0619 
So I was able to install TetWild in Ubuntu but the problem is that the object is not being able to be loaded as a softbody into pybullet:
1. It is taking more than 10 min to load (does the softbody implementation in pybullet has size limitation on objects?)
2. Also the .vtk file from gmsh is not loading - **Pybullet crashes or closes** when I am trying to load it, could you kindly see if there is any bug in pybullet implementation or is there any other reason for this behavior?

*  Here are the object files, as a .zip folder attached below, which I am trying to load as a softbody in pybullet:
.stl -> is the source file
.obj -> created using blender, like the files used in pybullet
.msh -> created using TetWild, following the comments above
.vtk -> created using gmsh export, like the files used in pybullet
[Sample-Spinner_Press_obj-SoftBodyTestPyb.zip](https://github.com/bulletphysics/bullet3/files/7659697/Sample-Spinner_Press_obj-SoftBodyTestPyb.zip)

renatyv, 2022-09-30:
I managed to generate valid .vtk meshes.
1) Blender: .obj -> .stl
2) TetWild: .stl -> .msh
3) gmsh: .msh -> .vtk
see [my post](https://medium.com/@renatyv/generating-vtk-mesh-from-for-pybullet-softbody-simulation-ccc5f2f39139) for more details

VallesMarinerisExplorer, 2023-07-18:
If any windows users are having trouble building tetwild for windows 10/possibly 11 there is a link to the binary here:   https://github.com/Yixin-Hu/TetWild/issues/47. Does anyone know how to create a command line argument for tetwild to convert stl to msh for instance?

VallesMarinerisExplorer, 2023-07-20:
This worked for me:

import gmsh
gmsh.initialize()

stl = gmsh.merge('Test.stl')
gmsh.model.mesh.classifySurfaces(gmsh.pi, True, True, gmsh.pi)
gmsh.model.mesh.createGeometry()

s = gmsh.model.getEntities(2)
surf = gmsh.model.geo.addSurfaceLoop([e[1] for e in s])
vol = gmsh.model.geo.addVolume([surf])
gmsh.model.geo.synchronize()
gmsh.model.occ.removeAllDuplicates()
gmsh.option.setNumber('Mesh.Algorithm', 1)
gmsh.option.setNumber('Mesh.MeshSizeMax', 5)
gmsh.option.setNumber('Mesh.MeshSizeMin', 0.5)

gmsh.model.addPhysicalGroup(3, [1], 1)
gmsh.model.setPhysicalName(3, 1, "The volume")
gmsh.model.occ.synchronize()
gmsh.model.mesh.generate(3)
gmsh.write("output.vtk")
gmsh.finalize()

REG1992, 2023-10-25:
> If any windows users are having trouble building tetwild for windows 10/possibly 11 there is a link to the binary here: [Yixin-Hu/TetWild#47](https://github.com/Yixin-Hu/TetWild/issues/47). Does anyone know how to create a command line argument for tetwild to convert stl to msh for instance?

Bump! I'm also struggling here.