bulletphysics/bullet3 #2728: Compute reward from state for DeepMimic environment?

e-271, 2020-04-08:
Hello,

We are using the HumanoidDeepBulletEnv in a model-based planning setting, and need to compute a reward based on a predicted state rather than on the current pose. It seems that calc_reward relies on the current pose from the pybullet engine and cannot do this. Is there another function we can use, or can we get some pointers how to convert the state into pose? Thanks.

Steven89Liu, 2020-04-10:
some ideas, hope this can help you.
1. refer to the updateCacheMultiDof() to caculate the joint pos based on the state.
2. refer to btMultiBody::compTreeLinkVelocities() to caculate the joint Vel based on the state.

erwincoumans, 2020-04-10:
This is Python code, did you try modifying it to do what you want?

e-271, 2020-04-12:
Actually, the part we are struggling with is how to calculate the position & velocity from the state. This seems to only be implemented in the pybullet engine currently.

erwincoumans, 2020-04-12:
The easiest is to make a copy in a new DIRECT sim, set that state and read the position and velocity from that sim.

erwincoumans, 2020-04-12:
please use the forums for this discussion.

closed 2020-04-12 by erwincoumans