bulletphysics/bullet3 #2729: Creating multi-bodies forces the orientation of the objects

sahandrez, 2020-04-08:
Hello, 

I'm getting a strange behavior from `createMultiBody`. When I add a falling object to the scene, it is somehow forced to reorient itself to the initial orientation of spawning, even long after it was spawned. 
This is a minimal example for reproducing it:
```
import pybullet as p
import pybullet_data
import time

def add_object(graphic_file="duck.obj",
               collision_file="duck_vhacd.obj",
               base_position=(0., 0., 0.),
               base_orientation=(0., 0., 0., 1.),
               mesh_scale=(1, 1, 1)):
    shift = [0, 0, 0]
    visualShapeId = p.createVisualShape(shapeType=p.GEOM_MESH,
                                        fileName=str(graphic_file),
                                        rgbaColor=[1, 1, 1, 1],
                                        specularColor=[0.4, .4, 0],
                                        visualFramePosition=shift,
                                        meshScale=mesh_scale)
    collisionShapeId = p.createCollisionShape(shapeType=p.GEOM_MESH,
                                              fileName=str(collision_file),
                                              collisionFramePosition=shift,
                                              meshScale=mesh_scale)

    obj_id = p.createMultiBody(baseMass=1,
                               baseInertialFramePosition=[0, 0, 0],
                               baseInertialFrameOrientation=[0, 0, 0, -1],
                               baseCollisionShapeIndex=collisionShapeId,
                               baseVisualShapeIndex=visualShapeId,
                               basePosition=base_position,
                               baseOrientation=base_orientation,
                               useMaximalCoordinates=False)
    return obj_id

if __name__ == '__main__':
    time_step = 1 / 240
    p.connect(p.GUI)
    p.setGravity(0, 0, -10)
    pybullet_data.getDataPath()
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    p.setTimeStep(time_step)
    p.loadURDF("plane100.urdf")

    for i in range(100):
        add_object(base_position=[0, 0, 1], mesh_scale=[0.1, 0.1, 0.1])
        for j in range(100):
            p.stepSimulation()
            time.sleep(time_step)
```
The script is basically spawning multiple objects at a height which of course fall on each other one by one. What I don't understand is that after they hit the ground, they unrealistically reorient to the baseOrientation. How can I disable that? 

Thank you.

erwincoumans, 2020-04-09:
This is due to the inertial center of mass frame position, Change the shift = [0, -0.08, 0]
(use the [createVisualShape.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/createVisualShape.py) script and change shift as above)

closed 2020-04-09 by erwincoumans

sahandrez, 2020-04-09:
Thank you. Could you please elaborate on the reason? Because now I should find a proper shift for any arbitrary object that I would want to add to the scene.

erwincoumans, 2020-04-09:
You claim to be a researcher, go research about the meaning of an inertial frame, essential in robotics.

sahandrez, 2020-04-09:
Right and changing the baseInertialFramePosition is the right action here. However. in the [createVisualShape.py(https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/createVisualShape.py) example, the visualFramePosition in createVisualShape is changed which according to the doc is the "translational offset of the visual shape with respect to the link frame" and is related only to the visual of the object (rather than dyanmics). I know software-wise they have the same outcome, but the latter seems unintuitive as we are modifying a "dynamics" property in a function for creating visual shapes.