bulletphysics/bullet3 #2730: C++ API

chris0914, 2020-04-09:
Hi Everyone,

I use PyBullet to simulate the turtlebot in a given environment, since I prefer it over Gazebo. Now I need a similar simulation in C++ and integrated the bullet3 project into my folder and everything compiles fine. The simulation with an empty world starts also fine, in DIRECt and GUI mode. However as soon as I load the turtlebot URDF, just two parts in wrong correlation are placed and no error returns. The file is fine, since its correctly displayed in PyBullet. Does anyone have any experience with the C++ API?

Cheers Chris

erwincoumans, 2020-04-09:
It is probably some build system issue. Bullet comes with an example how to use the C++ BulletRobotics API and it works just fine. See App_RobotSimulator (using premake or cmake).
Otherwise, please use the forums instead of this issue tracker.

closed 2020-04-09 by erwincoumans

chris0914, 2020-04-17:
First of all, thanks for the quick answer. That already helped me a lot. Indeed it was a build system issue and now the simulation runs fine, when created within my qt project. 
However, I managed to reproduce the error by creating a QApplication before starting the Simulation, resulting in the same parsing error as mentioned in my original post. It seems like all float values (mass, size etc.) are casted to integer. When the simulation is created first the robot gets parsed correctly. Any idea why QApplication has such effect?