bulletphysics/bullet3 #2734: How are objects in btDbvtBroadphase moved from the static to the dynamic tree?

Levi-Armstrong, 2020-04-10:
I am trying to understand in btDbvtBroadphase class how objects get move from the static tree to the dynamic tree and the other way around. It looks like to start all objects are added in the static tree and then when you update the aabb it moves it to the dynamic tree, but I am not sure. Also not how dynamic objects get moved from dynamic tree to static tree.

erwincoumans, 2020-04-10:
This is C++, are you familiar with it? You can add printf or use a debugger to see what is happening. Please use the forums to discuss.

closed 2020-04-10 by erwincoumans

Levi-Armstrong, 2020-04-10:
I do not understand the need to be rude. I am familiar C++ and also a maintainer of several open-source repositories.

erwincoumans, 2020-04-10:
Not intended to be rude, but I have no other answer then check out the source code or use the forums to understand more about this dynamic aabb broadphase. And more importantly, I don't have enough resources to reduce the number of real issues, and I still want to clear up this tracker, so I try to refer general questions to the forums.

OK, quick try:
1) check out Erin Catto's general overview of dynamic aabb tree:
https://box2d.org/files/ErinCatto_DynamicBVH_GDC2019.pdf

2) static objects (mass = 0) are always in the static tree
3) dynamic objects are either awake or sleeping. They start in the dynamic tree, and once they sleep they are moved to the 'static' tree (since we don't need to check sleeping/static objects against eachother).

4) adding and removing objects from a tree it a bit complicated, hence read Erin's doc and the source code, printf and debugger.

Levi-Armstrong, 2020-04-10:
Thank you, the information you provided is great! I will start reading the pdf you provided. In my application I using the low level broadphase and narrowphase components for collision checking during motion planning. I believe I may be incorrectly adding, removing and updating and the information you provided will be helpful in tracking down the issue. Also in the future I will post question on the forums.