bulletphysics/bullet3 #2737: Robot description not received in TCP client

missxa, 2020-04-10:
When connecting pybullet to a TCP server or trying to syncBodyInfo I get:
`b3Warning[examples/SharedMemory/PhysicsDirect.cpp,736]:
Robot description not received`. I cannot access robot's joints. What could be the issue?

erwincoumans, 2020-04-10:
Are they on the same platform? Compiled using 64bit double precision on both ends? Does the server have bodies? Please discuss further questions in the forums at https://pybullet.org

closed 2020-04-10 by erwincoumans

missxa, 2020-04-21:
is there a chance it's a bug in the implementation? or it's my lack of knowledge how to use the TCP implementation? i've described my issue in detail in a [post on the forum](https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=24&t=12926), but got no responces at all

the issue persists even on the same platform (with 64bit double precision) and i confirmed using shared memory connection that the server has bodies with joints...

erwincoumans, 2020-04-21:
There could be some issue(s) in the implementation.
Here are some fixes upcoming: https://github.com/bulletphysics/bullet3/pull/2757

missxa, 2020-04-22:
yup, those fixes solve the probem

erwincoumans, 2020-04-22:
Thanks for reporting!