bulletphysics/bullet3 #2738: malloc error with deployed libraries from standard bullet build in custom project

mPorst, 2020-04-11:
Hello, 
I am setting up a private cmake-based project with bullet.
Compilation of bullet works flawlessly via "build_cmake_pybullet_double.sh" and the app browser works. I installed the files into my project directory for linking. Deployed libraries are shared (!)
I then set up my project and attempted to reproduce the hello world steps. At the constructor of "btDiscreteDynamicsWorld" the program breaks and prints out something related to malloc which seems to tell me that it went out of range. (if you allocate the new object on the stack instead of the heap gcc will tell about stack smashing)
This guy seems to have had the same issue:
https://stackoverflow.com/questions/53323712/segmentation-fault-during-malloc-within-bullet-physics-library
From several other pages I found it seems that this problem occurs with gcc only but if used also on windows.

My system:
Arch Linux 5.5.13
g++ v 9.3.0
cmake version 3.17.1

Now **I found a "fix"** to this actually attempting to debug it. Building bullet using -DCMAKE_RELEASE_TYPE=DEBUG I received static libraries instead of (previously) shared libraries. The hello world project then worked as expected. I tested the same thing without any options and again received static libraries which - again - work as expected when I compile the hello world program. So it (probably) had not to do with compiler optimisations

I can't see anything in the build_cmake_pybullet_double.sh script that forces dynamic libraries and I have no knowledge of bullet's build-ecosystem so I figured it would be good to feed this info to the experts. 

**Steps to reproduce (linux):**
(get bullet sources)
./build_cmake_pybullet_double.sh
cd build_cmake
cmake --install --prefix /project-dir/bullet

check that deployed libraries are actually shared. Then set up minimal cmake-based project (with the HelloWorld.cpp source) that links with these deployed libraries. You should receive the same malloc error. I can provide you such a minimal project if you wish.
note that you have to make minimal changes to the BulletConfig.cmake file since it has some assumptions inside that are not true for this kind of deployment.

Best regards
Moritz

erwincoumans, 2020-04-11:
We only support static libraries, not shared libraries. 
This is not really a Bullet issue but some build system issue. Good luck.

closed 2020-04-11 by erwincoumans

mPorst, 2020-04-12:
I agree this is an issue with the build system. But shouldn't a "vanilla" build using the dedicated build script (build_cmake_pybullet_double.sh) create only the supported library versions, i.e. the static ones ? I mean you obviously support cmake ?

VasanthBalguri, 2022-06-02:
Think build_cmake_pybullet_double.sh is not good enough
Following standard way of cmake did the trick(using parameters)

mkdir build
cmake .. -DBUILD_SHARED_LIBS=OFF -> optional but helped for me, can add other parameters as well
make

Note: tweaking build_cmake_pybullet_double.sh by adding  -DBUILD_SHARED_LIBS=OFF doesn't seem to work, It seems to ignore these parameters for no reason.
Clear cmake cache, if any changes are made to library paths.