bulletphysics/bullet3 #2739: Stable PD Control with custom urdf model

LeeChunHei, 2020-04-12:
I am trying to edit the deepmimic RL with my own model.
However, when I try to control the model with stable PD control, all the joints did not respond at all.
I had read issue #2716 and understand that stable PD control is implemented with some assumption.
I had modified the source code from zeshiYang (#2716) with my model and added a fixed joint to the world like deepmimic's humanoid model and zeshiYang's model but it still acts the same.
Is there any other hidden requirement that my model did not fit to use stable PD control?

[image]
This is what it looks like when my model is using stable PD control, its target position is set to 0.

[humanoid.zip](https://github.com/bulletphysics/bullet3/files/4466326/humanoid.zip)
And this is the code that I use

erwincoumans, 2020-04-12:
deepmimic humanoid has a floating base, no fixed link to the world.

closed 2020-04-12 by erwincoumans

LeeChunHei, 2020-04-12:
> deepmimic humanoid has a floating base, no fixed link to the world.

Sorry but I still do not understand what a floating base should like.
from [deepmimic humanoid file](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/humanoid/humanoid.urdf) I saw these line
```
<robot name="dumpUrdf">
	<link name="base" >
		<inertial>
				<origin rpy = "0 0 0" xyz = "0 0 0" />
			 <mass value = "0.0001" />
			<inertia ixx = "0.0001" ixy = "0" ixz = "0" iyy = "0.0001" iyz = "0" izz = "0.0001" />
		</inertial>
	</link>
	<link name="root" >
		<inertial>
				<origin rpy = "0 0 0" xyz = "0.000000 0.280000 0.000000" />
			 <mass value = "6.000000" />
			<inertia ixx = "0.1" ixy = "0" ixz = "0" iyy = "0.1" iyz = "0" izz = "0.1" />
		</inertial>
		<collision>
				<origin rpy = "0 0 0" xyz = "0.000000 0.280000 0.000000" />
			<geometry>
				<sphere radius = "0.360000" />
			</geometry>
		</collision>
	</link>
	<joint name="root" type="fixed" >
		<parent link = "base" />
		<child link="root" />
				<origin rpy = "0 0 0" xyz = "0.000000 0.000000 0.000000" />
	</joint>
```
which it shows the deepmimic humanoid's root is connected to the world with a fixed joint.
Isn't this mean in order to use the stable PD control, I need to add a world link to the robot urdf file and connect the world link with a fixed joint to the robot's torso and at last import the files with useFixedBase=False?
But what I am facing now is even doing this, all the joints are still not driven.

erwincoumans, 2020-04-12:
base is not world.