bulletphysics/bullet3 #2741: bullet_robotics, bullet_robotics_gui

museum-future, 2020-04-12:
The release version does not install headers into

` include/bullet_robotics` 

or 
`include/bullet_robotics_gui`

when i correct this in the cmake files

```
  INSTALL( FILES ${BulletRoboticsGUI_INCLUDES}
           DESTINATION include/bullet_robotics_gui
  )
```

and

```
INSTALL (
    FILES ${BulletRobotics_INCLUDES}
    DESTINATION include/bullet_robotics
    )
```
, a warning  is shown

> CMake Warning (dev) at Extras/BulletRoboticsGUI/CMakeLists.txt:173 (INSTALL):
>   Target 'BulletRoboticsGUI' was changed to a FRAMEWORK sometime after
>   install().  This may result in the wrong install DESTINATION.  Set the
>   FRAMEWORK property earlier.
> This warning is for project developers.  Use -Wno-dev to suppress it

erwincoumans, 2020-04-21:
Are you using shared libraries on Mac?
There is this line that may conflict?
~~~~~~~~~~~~~~~~~
SET_TARGET_PROPERTIES(BulletRobotics  PROPERTIES FRAMEWORK true)
~~~~~~~~~~~~~~~~~
If you have a small patch, please consider a pull request.

erwincoumans, 2020-04-22:
Please coordinate with https://github.com/bulletphysics/bullet3/issues/2758 and provide a pull request of changes in BulletRobotics and BulletRoboticsGUI to copy/include the right headers in the right place for the platforms you care about (Windows, Mac, Linux etc)

museum-future, 2020-04-28:
Sorry for late response, I tried with and without shared libs. I can patch it for OSX.

closed 2020-04-28 by museum-future