bulletphysics/bullet3 #2746: Fixing seeds in pybullet environment HalfCheetahBulletEnv-v0 does not work

rdednl, 2020-04-15:
`env.seed()` doesn't properly fix the sources of randomness in the environment `HalfCheetahBulletEnv-v0`. Running multiple times the same environment with the same seed doesn't produce same results.

### Code sample to reproduce behaviour:

```
import pybullet_envs
import gym
import numpy as np

for i in range(20):
    env=gym.make('HalfCheetahBulletEnv-v0')
    
    env.seed(0)
    env.action_space.seed(0)
    env.observation_space.seed(0)
    
    env.reset()
    
    total_reward = 0
    terminal = False
    
    while not terminal:
        state, reward, terminal, _ = env.step(np.zeros(env.action_space.shape))
        total_reward += reward

    print(total_reward)
```

Expected behavior is 20 numbers (`total_reward`) to be exactly the same. Result instead fluctuates between -1353 and -1359.
It happens sometimes that all the 20 numbers are the same but it is not consistent.

[image]

erwincoumans, 2020-04-15:
Which platform/version, pybullet and python version?
When I run the script on Windows, Python 3.8, I get this:
~~~~~~~~~~~~~
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
-1353.9260731497207
~~~~~~~~~~~~~

rdednl, 2020-04-15:
Seeding numpy shouldn't be needed (as it shouldn't be needed seeding the action space and observation space as we are not using randomness there as far as I know). I just tried seeding numpy and it still behaves in the same way. I managed to reproduce the same inconsistent result also with `Walker2DBulletEnv-v0`.

Has been tried on:
1. Ubuntu 18.04.4, python 3.6.9, gym 0.15.4, pybullet 2.6.4
2. Ubuntu 18.04.4, python 3.8.1, gym 0.17.1, pybullet 2.6.5
3. MacOS 10.15.3, python 3.7.3, gym 0.15.6, pybullet 2.6.4

I will report back after trying on a Windows machine too.

erwincoumans, 2020-04-15:
That is a very old pybullet version, can you please use the latest? 2.7.3 as of now.

closed 2020-04-15 by erwincoumans

rdednl, 2020-04-15:
Tested on the new version and it's fixed. Thanks for closing it.

OlivierBelan, 2023-11-21:
Hello,

I'm encountering an issue similar to what has been discussed in this thread. I'm using `pybullet==3.2.5` alongside `gym==0.26.2`, and I've also tried with `gymnasium==0.28.1` (updating imports accordingly). The problem arises specifically with the HalfCheetah environment in PyBullet.

Interestingly, this issue seems exclusive to PyBullet. When I switch to the HalfCheetah environment from Gymnasium, the seed reset functions as expected.

Here's how I'm setting the seed in PyBullet:

```python
self.gym_env.seed(seed)
self.gym_env.action_space.seed(seed)
self.gym_env.observation_space.seed(seed)
self.observation = self.gym_env.reset()
```

And here's the approach for Gymnasium:

```python
self.observation, self.extra_info = self.gym_env.reset(seed=self.seed)
```

My setup is on Ubuntu 22, and I'm using Python 3.10.12. Any insights or suggestions to resolve this inconsistency with PyBullet would be greatly appreciated.