bulletphysics/bullet3 #2747: PPO train kuka_grasping didn't perform well

LINGYUNFDU, 2020-04-15:
I have tried:
python3 -m pybullet_envs.agents.train_ppo --config=pybullet_kuka_grasping --logdir=kuka
Then i evaluate the environment:
python3 -m pybullet_envs.agents.visualize_ppo  --logdir=kuka/<timestampname> --outdir=kuka_video1

However, from the "kuka_video1", we find it the robot can still not grasp the object.

_Originally posted by @pyni in https://github.com/bulletphysics/bullet3/issues/1386#issuecomment-384849069_

52Hz-Kun, 2020-04-19:
Hello, I'd like to ask you if you have successfully used dqn algorithm experiment?

LINGYUNFDU, 2020-04-20:
> Hello, I'd like to ask you if you have successfully used dqn algorithm experiment?

No, neither the DQN on discrete grasping task nor this PPO on continus task perform well... sigh

erwincoumans, 2020-04-21:
It is not supported. Please find a forum to discuss it.

closed 2020-04-21 by erwincoumans