bulletphysics/bullet3 #2751: PyBullet: Connection mode influences rendering of Box geoms in MJCF files

jgehring, 2020-04-17:
It appears that the connection mode modifies how bodies are loaded from MJCF -- the symptom is that their camera image rendering is affected. See [this script](https://gist.github.com/jgehring/95d42638454b016cc9cf434b552a1952) for reproducing this (tested with PyBullet 2.7.2 on macOS and 2.7.3 on Ubuntu). The script creates 3 bodies: a box and a sphere from a MCJF file, and a equal-sized box via `createMultiBody()`. Colors defined in MCJF descriptions don't seem to be recognized, but box geoms are rendered at a significantly smaller size when connecting in `DIRECT` mode; in `GUI` mode, shapes are as expected. Interestingly, the mass and extent of the MCJF box when queried via the API is the same in both connection modes.

Here are the images produced by the script:

`p.connect(p.GUI)` | `p.connect(p.DIRECT)`
------------ | -------------
[image] | [image]
[image] | [image]

jgehring, 2020-04-17:
Update: in the example script above, I used `ER_BULLET_HARDWARE_OPENGL` to obtain the camera image since I copy-pasted the rendering code. When I use `ER_TINY_RENDERER`, I obtain the right-hand column from above (i.e. center box rendered too small) for both connection modes. Apparently, it's the rendering back-end rather than the connection mode that results in invalid box sizes.

erwincoumans, 2020-04-18:
Yes, tinyrenderer converted from urdf convention (urdf size is actual size, while mjcf size is half-size)
fixed here. Most of the standard mjcf benchmarks (ant, hopper, half-cheetah, humanoid only uses sphere and capsules, so it went undetected).
Thanks for the report!

closed 2020-04-18 by erwincoumans