bulletphysics/bullet3 #2753: Failed building wheel for pybullet

Mister3119, 2020-04-19:
pip3 install pybullet --upgrade --user
pip3 install pybullet
pip install pybullet

not work

erwincoumans, 2020-04-19:
It does for most people, you must be a snowflake. Please share your magic fairy dust: all versions of all important parts, os, compiler, python.

closed 2020-04-19 by erwincoumans

Mister3119, 2020-04-19:
j'ai deux pc un sa marche mais pas l'autre mais sa marche sur un pc portable pas puissant du tout j'ai toute les dernière version j'ai tester de réinstaller mais marche toujours pas je suis bloqué 
[erreur.txt](https://github.com/bulletphysics/bullet3/files/4500041/erreur.txt)

erwincoumans, 2020-04-19:
The most important part is missing: what version of visual studio did you install? It is best to use Visual Studio 2019 and run pip install inside a 'X64 Native Tools command prompt for VS2019' terminal.