bulletphysics/bullet3 #2755: Issue with Remote UDP Physics Server

ZhiquanW, 2020-04-20:
According to 'Pybullet Quickstart Guide', there is a way to connect to the physics server over UDP tunneling. But I did not see any details about that. Is there any examples or will the details be released in the furture?
I do spend a lot of time looking for a way to establish a UDP/TCP server but i failed. The quickstart guide only provides the API to connect to UCP/TCP server, which is weired.

[image]

erwincoumans, 2020-04-21:
You will have to compile the physics server tcp bridge from c++ at the moment, it is not built-in.
I'll embed it at some stage.
There is a graphics server embedded instead, over TCP. See 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/graphicsClient.py
and
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/graphicsServer.py

closed 2020-04-21 by erwincoumans

Mridul1129261, 2023-10-03:
I cannot load URDF from pybullet.GRAPHICS_SERVER on device 1. However, I can load URDF from the pybullet.GRAPHICS_SERVER_TCP on device 2 to connect to device 1.
Is there any way for me to be able to load URDF at a single server from two different devices connected over the TCP or UDP?

Device 1 code-

`import pybullet as p
import time
import pybullet_data
p.disconnect() if p.isConnected() else print("Ready")
pid=p.connect(p.GRAPHICS_SERVER)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
print("started graphics server",pid)
p.loadURDF("plane.urdf")                         #this line throws error - cannot load URDF files`

Device 2 code-

`import pybullet as p
import pybullet_data as pd
import time
ip_addr = 192.xxx.xx.x                                               #ip v4 address of device 1
p.connect(p.GRAPHICS_SERVER_TCP,ip_addr,6667)
p.setAdditionalSearchPath(pd.getDataPath())
p.syncBodyInfo()
print(p.getNumBodies())
p.loadURDF("r2d2.urdf", [1,1,3])
p.setGravity(0,0,-10)
if p.isConnected():
  for _ in range(100):
    p.stepSimulation()
    time.sleep(0.01)`
    
I am using Windows 11 x64 operating system and VScode version 1.82.3