bulletphysics/bullet3 #2758: Using the BulletRobotics library with GUI

aravindsiv, 2020-04-22:
Hi,

I've been using the BulletRobotics API (present inside Extras) in my own project, where I install it using CMake by adding the `BulletRobotics` library to `BULLET_LIBRARIES`. While this works great, I'd like to use the GUI as well, which is not possible with the currently built BulletRobotics library, since it does not use `b3RobotSimulatorClientAPI.h` as a source (only the NoGUI and NoDirect versions of the API are used). I tried adding it as a source, but it seems to have multiple dependencies of its own. 

I was wondering if there was a cleaner/easier fix for this than trying to locate all the dependencies and adding them to `BulletRobotics_SRCS` inside BulletRobotics's CMakeLists.txt. 

Thanks in advance!

erwincoumans, 2020-04-22:
There is the Extras/BulletRoboticsGUI library for that purpose, did you try it?

aravindsiv, 2020-04-22:
I'm using the latest release (2.89) which doesn't have the Extras/BulletRoboticsGUI library. I'll try to copy the CMake file for that library and see if it works.

aravindsiv, 2020-04-22:
Not sure if my edited comment above triggers a notification (if so, sorry for the spam):

I tried to copy the CMake file for the Extras/BulletRoboticsGUI library, but it doesn't install b3RobotSimulatorClientAPI.h into my project's include folder. It is not even present in BulletRoboticsGUI_INCLUDES in the CMakeLists. It does install libBulletRoboticsGUI.so into my project's lib folder. I'm confused as to how to import the b3RobotSimulatorClientAPI class in my project.

erwincoumans, 2020-04-22:
CMake evolves with community contributions. Can you coordinate here to make it work and provide a pull request?
https://github.com/bulletphysics/bullet3/issues/2741

erwincoumans, 2020-05-01:
Using a plain install of Bullet will install those libraries:
~~~~~~~
libBullet2FileLoader.a         
libBullet3Collision.a          
libBullet3Common.a             
libBullet3Dynamics.a           
libBullet3Geometry.a           
libBullet3OpenCL_clew.a        
libBulletCollision.a           
libBulletDynamics.a            
libBulletExampleBrowserLib.a   
libBulletFileLoader.a          
libBulletInverseDynamics.a     
libBulletInverseDynamicsUtils.a
libBulletRobotics.a            
libBulletRoboticsGUI.a         
libBulletSoftBody.a            
libBulletWorldImporter.a       
libBulletXmlWorldImporter.a    
libBussIK.a                    
libclsocket.a                  
libConvexDecomposition.a       
libGIMPACTUtils.a              
libgwen.a                      
libHACD.a                      
libLinearMath.a                
libOpenGLWindow.a              
~~~~~~~

You just have to re-implement or copy b3RobotSimulatorClientAPI.cpp into your own project and it should all be fine.

closed 2020-05-01 by erwincoumans

aravindsiv, 2020-06-29:
Hi @erwincoumans ,

I'm facing an issue when linking the libraries. For reference, my `CMake` and `make` commands inside `build_cmake_pybullet_double.sh` are:

```
cmake -E env CXXFLAGS="-fPIC" cmake -DBUILD_PYBULLET=OFF -DBUILD_PYBULLET_NUMPY=OFF DUSE_DOUBLE_PRECISION=ON -DBT_USE_EGL=ON -DCMAKE_INSTALL_PREFIX=/path/to/app/ .. || exit 1
make install -j $(command nproc 2>/dev/null || echo 12) || exit 1
```

When I am linking the libraries in my application, I get the following error message:

```
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::unloadPlugin(int)':
b3PluginManager.cpp:(.text+0xdc7): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::registerStaticLinkedPlugin(char const*, b3PluginFunctions&, bool)':
b3PluginManager.cpp:(.text+0x164c): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x2e12): undefined reference to `dlopen'
b3PluginManager.cpp:(.text+0x356a): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x357e): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x3592): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x35a6): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x35ba): undefined reference to `dlsym'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o):b3PluginManager.cpp:(.text+0x35ce): more undefined references to `dlsym' follow
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x36f2): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x3a30): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x3ab6): undefined reference to `dlerror'
b3PluginManager.cpp:(.text+0x3e42): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `threadFunction(void*)':
b3PosixThreadSupport.cpp:(.text+0x2c): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x42): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x56): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x94): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0xa8): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x10c): undefined reference to `sem_post'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::waitForResponse(int*, int*)':
b3PosixThreadSupport.cpp:(.text+0x158): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x1d6): undefined reference to `sem_wait'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::isTaskCompleted(int*, int*, int)':
b3PosixThreadSupport.cpp:(.text+0x252): undefined reference to `sem_trywait'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::runTask(int, void*, int)':
b3PosixThreadSupport.cpp:(.text+0x30c): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x32c): undefined reference to `sem_post'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::createBarrier()':
b3PosixThreadSupport.cpp:(.text+0x3a8): undefined reference to `pthread_barrier_init'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::deleteBarrier(b3Barrier*)':
b3PosixThreadSupport.cpp:(.text+0x44c): undefined reference to `pthread_barrier_destroy'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::stopThreads()':
b3PosixThreadSupport.cpp:(.text+0x495): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x4b5): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x4c9): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x50d): undefined reference to `pthread_join'
b3PosixThreadSupport.cpp:(.text+0x555): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x56e): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x59c): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x5b5): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x5f7): undefined reference to `pthread_join'
b3PosixThreadSupport.cpp:(.text+0x63c): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x650): undefined reference to `sem_destroy'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::startThreads(b3PosixThreadSupport::ThreadConstructionInfo&)':
b3PosixThreadSupport.cpp:(.text+0x72e): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x7f4): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x811): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x854): undefined reference to `pthread_create'
b3PosixThreadSupport.cpp:(.text+0x87d): undefined reference to `pthread_create'
b3PosixThreadSupport.cpp:(.text+0x8d0): undefined reference to `sem_init'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixBarrier::setMaxCount(int)':
[... cut ...]
egl.c:(.text+0x1146): undefined reference to `dlsym'
egl.c:(.text+0x1163): undefined reference to `dlsym'
egl.c:(.text+0x1180): undefined reference to `dlsym'
egl.c:(.text+0x119d): undefined reference to `dlsym'
/path/to/app/lib/libOpenGLWindow.a(egl.o):egl.c:(.text+0x11ba): more undefined references to `dlsym' follow
/path/to/app/lib/libOpenGLWindow.a(egl.o): In function `gladLoaderUnloadEGL':
egl.c:(.text+0x1a81): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
```

Any idea what might be going wrong?