bulletphysics/bullet3 #2760: StrikerBulletEnv-v0 is not working

pengzhenghao, 2020-04-23:
```python
import pybullet_envs
import gym

e = gym.make("StrikerBulletEnv-v0")
e.reset()
```

Raise error:

```
~/anaconda3/envs/dev/lib/python3.7/site-packages/pybullet_envs/robot_bases.py in reset(self, bullet_client)
    139         self.parts, self.jdict, self.ordered_joints, self.robot_body = self.addToScene(
    140             self._p, self.objects)
--> 141     self.robot_specific_reset(self._p)
    142
    143     s = self.calc_state(

~/anaconda3/envs/dev/lib/python3.7/site-packages/pybullet_envs/robot_manipulators.py in robot_specific_reset(self, bullet_client)
    147     # parts
    148     self.fingertip = self.parts["fingertip"]
--> 149     self.target = self.parts["target"]
    150     self.object = self.parts["object"]
    151

KeyError: 'target'
```

erwincoumans, 2020-04-23:
Yes, we better remove the broken env (the original contributor is doing other stuff now, and I don't have time to fix that). Unless you feel like fixing it, we can remove the env from tne __init__ file in bullet3/examples/pybullet/gym/pybullet_envsac

erwincoumans, 2020-04-23:
Disabled here. 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/__init__.py
If you find more broken envs, please either fix them or disable them in the __init__

closed 2020-04-23 by erwincoumans