bulletphysics/bullet3 #2763: Cannot compile bullet in Ubuntu 19.10 docker with Gcc 9

qarmin, 2020-04-23:
When I use this configuration in .gitlab-ci.yml
```
    - apt install -y make gcc g++ libx11-dev x11proto-dev
    - cd build3
    - ./premake4_linux64 --double gmake
    - cd gmake
    - make -j6
```
Then at the end of compilation, this errors will be shown(strange  that it works ~1 month ago):
```
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp: In member function 'void BatchRayCaster::processRay(int)':
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:6051:10: error: 'btSoftBody' does not name a type; did you mean 'btSolverBody'?
 6051 |    const btSoftBody* softBody = btSoftBody::upcast(rayResultCallback.m_collisionObject);
      |          ^~~~~~~~~~
      |          btSolverBody
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:6052:16: error: 'softBody' was not declared in this scope
 6052 |    if (body || softBody)
      |                ^~~~~~~~
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp: In member function 'bool PhysicsServerCommandProcessor::processDeformable(const UrdfDeformable&, const btVector3&, const btQuaternion&, int*, char*, int, btScalar, bool)':
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8306:2: error: 'btSoftBody' was not declared in this scope; did you mean 'btSolverBody'?
 8306 |  btSoftBody* psb = NULL;
      |  ^~~~~~~~~~
      |  btSolverBody
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8306:14: error: 'psb' was not declared in this scope
 8306 |  btSoftBody* psb = NULL;
      |              ^~~
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8354:13: error: 'btSoftBodyHelpers' has not been declared
 8354 |       psb = btSoftBodyHelpers::CreateFromTriMesh(softWorld->getWorldInfo(), &vertices[0], &indices[0], numTris);
      |             ^~~~~~~~~~~~~~~~~
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8354:59: error: invalid use of incomplete type 'class btSoftMultiBodyDynamicsWorld'
 8354 |       psb = btSoftBodyHelpers::CreateFromTriMesh(softWorld->getWorldInfo(), &vertices[0], &indices[0], numTris);
      |                                                           ^~
In file included from ../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:1:
../../examples/SharedMemory/PhysicsServerCommandProcessor.h:26:8: note: forward declaration of 'class btSoftMultiBodyDynamicsWorld'
   26 |  class btSoftMultiBodyDynamicsWorld* getSoftWorld();
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8366:13: error: 'btSoftBodyHelpers' has not been declared
 8366 |       psb = btSoftBodyHelpers::CreateFromTriMesh(deformWorld->getWorldInfo(), &vertices[0], &indices[0], numTris);
      |             ^~~~~~~~~~~~~~~~~
../../examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8366:61: error: invalid use of incomplete type 'class btDeformableMultiBodyDynamicsWorld'
 8366 |       psb = btSoftBodyHelpers::CreateFromTriMesh(deformWorld->getWorldInfo(), &vertices[0], &indices[0], numTris);
      |                                                             ^~
```
Failed job - https://gitlab.com/qarmin/bullet3/-/jobs/523943277
Giltab-ci file - https://gitlab.com/qarmin/bullet3/-/blob/master/.gitlab-ci.yml

erwincoumans, 2020-04-23:
It is a bug, we only compile btSoftBody if not defined SKIP_SOFT_BODY_MULTI_BODY_DYNAMICS_WORLD

erwincoumans, 2020-04-23:
fixed here:
https://github.com/bulletphysics/bullet3/pull/2768

closed 2020-04-27 by qarmin