bulletphysics/bullet3 #2765: PyBullet: getContactPoints() API

jgehring, 2020-04-23:
I've been experimenting with the `getContactPoints()` API and have some questions (and a possible bug report) regarding it:

- If I call `getContactPoints()` without arguments (i.e. without filters), I get the body with the numerically smaller ID reported as bodyA and the one with the larger ID reported as bodyB. Is this something that can always be assumed, i.e. are the internal structures maintained in a sorted manner?
- If I swap arguments for bodyA/linkA and bodyB/linkB, I get the same return values for a contact, except for the body and link indices. Here's an example from one of my use-cases:
```py
ipdb> p.getContactPoints(bodyA=1, bodyB=2)
((0, 1, 2, -1, 3, (-0.011672196297677972, 0.011672196297677972, 0.3460310343882662), (-0.010370589270642994, 0.010370589270642994, 0.3626564612012315), (0.0778146419845198, -0.0778146419845198, 0.9939264374115632), -0.01672701941228381, 191.34750985763412, -41.134521057449426, (0.0, -0.9969493435839967, -0.0780513057382009), -40.88468796926162, (0.9969678437606807, 0.0060735344124424, -0.07757725624769073)),)
ipdb> p.getContactPoints(bodyA=2, bodyB=1)
((0, 2, 1, 3, -1, (-0.011672196297677972, 0.011672196297677972, 0.3460310343882662), (-0.010370589270642994, 0.010370589270642994, 0.3626564612012315), (0.0778146419845198, -0.0778146419845198, 0.9939264374115632), -0.01672701941228381, 191.34750985763412, -41.134521057449426, (0.0, -0.9969493435839967, -0.0780513057382009), -40.88468796926162, (0.9969678437606807, 0.0060735344124424, -0.07757725624769073)),)
```
Digging a bit further, I found [some code](https://github.com/bulletphysics/bullet3/blob/5ac86c5f68bef97c81e06e42a1b39f4e6062f476/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L7652-L7675 ) for swapping the body and link indices (and not the positions and force vectors) if a filter matches. I'm not sure if that's a relevant source code location, though.
- If I set up the simulation to use `numSubSteps` larger than 1, it appears like `getContactPoints()` only returns the contact information of the last simulation step. Is there any way to reliably query all collisions that occured within a single call to `stepSimulation()` regardless of `numSubSetps`?

erwincoumans, 2020-04-23:
That code is a bug indeed, thanks for reporting! That whole swap thing should be needed, you optionally provide two body unique ids and link indices, and the contact point can be in any order.

I wouldn't rely on the first index in a contact being smaller than the second.

>> only returns the contact information of the last simulation step.

Indeed, getContactPoints only reports the last simulation sub step. You would have to write a pybullet C-plugin to gather all points in a post-tick callback, or set sub-steps to 1 and manually call the simulation with smaller time steps.

erwincoumans, 2020-04-23:
Can you confirm this fixes the swap issue?
https://github.com/bulletphysics/bullet3/pull/2766
Thanks!

jgehring, 2020-04-23:
Thanks for the fast reply! The patch works well, yay :)

Thanks for the extra info regarding `getContactPoints`. For now, I'm going with a small client wrapper (in Python) that can pool contact information across multiple simulation steps; simply retaining the contact with maximum normal force per body/link pair should be enough for me at this point.

Just to confirm, if I would manually apply the `numSubSetps` logic in a control setting, I should apply the motor forces for each sub-step, i.e. before each call to a single-substep `stepSimulation()`, right? Based on a few of experiments this seems to yield identical behavior at least.

erwincoumans, 2020-04-23:
>> Just to confirm

Yes, if you use torque control you have to apply those torques each stepSimulation. 
(for setJointMotorControl2 in velocity or position control that is not needed)

jgehring, 2020-04-23:
Great, thanks!

closed 2020-04-23 by erwincoumans