bulletphysics/bullet3 #2770: TINY_RENDERER: Color Issue

chris0914, 2020-04-24:
Hi,

I am using the C++ API b3RobotSimulatorClientAPI. I render a mobile camera with the getCameraImage. I change the colors of my world with changeVisualShape, which works since the color is green/yellow randomly otherwise. The GUI shows everything correctly, as well as the image when rendered with ER_BULLET_HARDWARE_OPENGL. When ER_TINY_RENDERER is used everything, but one obstacle, is green (see screenshots). Unfortunately I am bounded to ER_TINY_RENDERER, because I want to use DIRECT mode.

Left: GUI, Right: getCameraImage, Top: OpenGL, Bottom: TinyRenderer

[image]

[image]

Kind regards,
Chris

aravindsiv, 2020-04-24:
Hi,

I saw your issue #2730 . Could you tell me how you were able to compile the C++ API as an external library with the GUI?

Thanks in advance!

erwincoumans, 2020-04-24:
Also check out the pybullet [cmake target](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/CMakeLists.txt), it depends on BulletRoboticsGUI etc.
~~~~~~~~~~~~~~
TARGET_LINK_LIBRARIES(pybullet BulletRoboticsGUI BulletExampleBrowserLib BulletRobotics BulletFileLoader BulletWorldImporter BulletSoftBody BulletDynamics BulletCollision BulletInverseDynamicsUtils BulletInverseDynamics LinearMath OpenGLWindow gwen BussIK Bullet3Common)
~~~~~~~~~~~~~~

Another option is to just depend on BulletRobotcs and manually add a few files. See 
https://github.com/bulletphysics/bullet3/blob/master/examples/RobotSimulator/CMakeLists.txt

chris0914, 2020-04-26:
Thanks for looking into it!

@aravindsiv 
I went the second option, similar to App_RobotSimulator. However I rewrote the main into a class and compiled it as a Library instead of an Executable. See  https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/CMakeLists.txt, as example.

erwincoumans, 2020-05-08:
I can't reproduce this. Did you use changeVisualShape for all links and base (linkIndex=-1)

~~~~~~~~~~~~~~~
import pybullet as p
import time

p.connect(p.GUI)
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0)
castle = p.loadURDF("samurai.urdf")
p.changeVisualShape(castle,-1,rgbaColor=[1,1,1,1])
for j in range(p.getNumJoints(castle)):
  p.changeVisualShape(castle,j,rgbaColor=[1,1,1,1])
  
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,1)
while (1):
  p.getCameraImage(320,200)
  p.stepSimulation()
  time.sleep(1./240.)
  
~~~~~~~~~~~~~~~

Please re-open if you have a repro case with small python script and all assets in a zipfile.

chris0914, 2020-05-11:
Hi erwincoumans,

as mentioned I am using the C++ API. I can't reproduce it in python though. However in C++ I have a MWE, I just cloned the newest bullet master and still get the error. 
To use it replace the file https://github.com/bulletphysics/bullet3/blob/master/examples/RobotSimulator/RobotSimulatorMain.cpp and the corresponding CMakeLists.txt and compile using  './build_cmake_pybullet_double.sh'. OpenCV is required to open the second window for the camera view and already added in the zipped CMakeLists.

[MWE.zip](https://github.com/bulletphysics/bullet3/files/4609665/MWE.zip)

Thanks in advance

chris0914, 2020-05-11:
@erwincoumans somehow I cannot re-open this issue. Please re-open for me

erwincoumans, 2020-06-01:
createVisualShape returns a shape unique id (the naming shapeIndex inside the code is wrong)
For changeVisualShape you should not use this shape unique id. You only provide an object unique id and a link index (-1 for base, 0 onwards for child links).
The 'shape index' only makes sense if you have multiple visual shapes for a specific link, and this shape index starts from zero for each link.

~~~~~~~~~~~~~~~~
 b3RobotSimulatorChangeVisualShapeArgs colorArgs;
        colorArgs.m_objectUniqueId = objID;
        //colorArgs.m_shapeIndex = 0;// nWallVis; //don't use nWallVis!
        colorArgs.m_hasRgbaColor = true;
        colorArgs.m_rgbaColor = col;
        sim->changeVisualShape(colorArgs);
~~~~~~~~~~~~~~~~

closed 2020-06-01 by erwincoumans

chris0914, 2020-06-02:
Thanks, works all great now!