bulletphysics/bullet3 #2772: Image generated by render(mode="rgb_array") cannot track the character well

tianxintao, 2020-04-25:
Hi,

I am using the `AntBulletEnv-v0`  implemented in pybullet_envs, and I am generating the images by calling `render(mode="rgb_array")`. But it seems that at the beginning, the camera keeps track of the ant well, but after a while, the character drift towards the boundary, then eventually, it is partially cut off by the boundary of the image. The code I attached below is exactly the same as one of the examples published in pybullet_envs, except rendering the image using `render(mode="rgb_array")`.

At the beginning:
[image]

Near the end of the episode
[image]

Do you have any idea why this happens? Any help is really appreciated!

python: 3.7.7
pybullet: 2.7.3
gym: 0.17.1

Code to reproduce this:
`#add parent dir to find package. Only needed for source code build, pip install doesn't need it.
import os
import inspect
import pybullet
import gym
import numpy as np
import pybullet_envs
import time
import cv2

def relu(x):
  return np.maximum(x, 0)

class SmallReactivePolicy:
  "Simple multi-layer perceptron policy, no internal state"

  def __init__(self, observation_space, action_space):
    assert weights_dense1_w.shape == (observation_space.shape[0], 128)
    assert weights_dense2_w.shape == (128, 64)
    assert weights_final_w.shape == (64, action_space.shape[0])

  def act(self, ob):
    x = ob
    x = relu(np.dot(x, weights_dense1_w) + weights_dense1_b)
    x = relu(np.dot(x, weights_dense2_w) + weights_dense2_b)
    x = np.dot(x, weights_final_w) + weights_final_b
    return x

def main():
  pybullet.connect(pybullet.DIRECT)
  env = gym.make("AntBulletEnv-v0")
  # env.render(mode="human")

  pi = SmallReactivePolicy(env.observation_space, env.action_space)
  env.reset()

  while 1:
    frame = 0
    score = 0
    restart_delay = 0
    obs = env.reset()

    while 1:
      # time.sleep(1. / 60.)
      a = pi.act(obs)
      obs, r, done, _ = env.step(a)
      #print("reward")
      #print(r)
      score += r
      frame += 1
      distance = 5
      yaw = 0

[... 2087 lines cut ...]
     [-0.2729, +0.0973, +0.2516, -0.4313, -0.1743, -0.2386, -0.1142, -0.0440],
     [-0.6820, +0.2449, -0.1079, +0.3047, +0.0325, -0.0980, +0.1868, -0.1600],
     [-0.2085, -0.2897, -0.4539, -0.0506, +0.1225, +0.1976, -0.2360, +0.6553],
     [+0.5299, +0.2876, +0.2354, -0.0884, -0.1821, +0.3841, -0.0701, +0.0278],
     [-0.1594, -0.4837, +0.1483, -0.1280, +0.0962, -0.1932, +0.1730, +0.0544],
     [+0.2879, -0.1846, +0.1474, -0.3383, +0.1483, +0.1159, +0.3761, +0.2326],
     [-0.0960, -0.0727, -0.3099, +0.0548, +0.2339, -0.2232, +0.6066, +0.3826],
     [-0.0171, +0.1490, -0.3324, -0.0231, +0.0700, -0.6364, -0.1944, +0.3094],
     [+0.0804, -0.0278, +0.0182, -0.3807, -0.3628, -0.3564, +0.2279, -0.0226],
     [+0.4219, +0.0741, +0.1299, +0.4700, +0.1927, +0.0849, -0.1574, +0.5607],
     [+0.0810, -0.0815, +0.1356, +0.2048, -0.1404, -0.0222, -0.2464, +0.4991],
     [+0.3000, +0.0862, -0.3917, +0.4117, +0.0306, +0.0788, -0.0403, -0.6423],
     [-0.2445, +0.4847, +0.0478, -0.2170, -0.3700, +0.0189, +0.0001, -0.0335],
     [+0.3710, -0.0115, -0.4207, -0.4245, +0.2041, -0.0875, -0.2369, -0.6424],
     [-0.3608, +0.0539, -0.3928, -0.0352, +0.2832, +0.1003, -0.0082, +0.3201],
     [+0.2221, -0.0399, +0.2092, -0.1670, -0.5146, +0.2273, +0.3011, -0.2879],
     [+0.1500, +0.3236, +0.1295, -0.0763, -0.2708, +0.6542, +0.1379, -0.0646],
     [+0.1291, +0.0417, +0.1386, +0.0166, +0.1373, +0.1956, +0.4986, +0.5690]])

weights_final_b = np.array(
    [-0.0680, +0.1401, -0.0628, -0.1317, +0.1489, +0.1844, -0.1147, +0.0137])
# yapf: enable

if __name__ == "__main__":
  main()`

tianxintao, 2020-04-25:
It seems that changing [line45](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/robot_locomotors.py#L45) in `robot_locomotors.py` to `self.body_xyz = body_pose.xyz()` could solve this problem. But still quite confusing, why the default setup cannot track the com well?

erwincoumans, 2020-04-25:
Those environments are messy code I'm not very familiar with. Please consider a pull request if you fixed it.

sgillen, 2020-04-25:
I think merging #2771 (Either into master or just your local branch @tianxintao), and then adding an env.unwrapped.camera_adjust() before the call to render will fix your issue. That's currently what I'm doing to resolve what I believe is the same issue in my own work.

tianxintao, 2020-04-25:
@sgillen Sounds promising. Thanks. Will try this

erwincoumans, 2020-04-26:
Sounds good, I'll merge the #2771 and add env.unwrapped.camera_adjust() to render.

erwincoumans, 2020-04-26:
There were a few issues that I fixed here:
https://github.com/bulletphysics/bullet3/pull/2768
render('human') didn't track objects, now fixed
render('rgb_array') lagged behind object

tianxintao, 2020-05-03:
Thank you for fixing this! However, in the latest version 2.7.5 where this issue is fixed, calling `env.render(mode="rgb_array")` gives me an error of:
```
requestOpenGLVisualizeCamera failedTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lustre03/project/6004621/tianxin/penv/lib/python3.7/site-packages/gym/core.py", line 235, in render
    return self.env.render(mode, **kwargs)
  File "/lustre03/project/6004621/tianxin/penv/lib/python3.7/site-packages/pybullet_envs/env_bases.py", line 99, in render
    self.camera_adjust()
  File "/lustre03/project/6004621/tianxin/penv/lib/python3.7/site-packages/pybullet_envs/gym_locomotion_envs.py", line 132, in camera_adjust
    self.camera.move_and_look_at(self.camera_x, y , 1.4, x, y, 1.0)
  File "/lustre03/project/6004621/tianxin/penv/lib/python3.7/site-packages/pybullet_envs/env_bases.py", line 169, in move_and_look_at
    camInfo = self.env._p.getDebugVisualizerCamera()
pybullet.error: Cannot get OpenGL visualizer camera info.
```

erwincoumans, 2020-05-05:
It also runs in DIRECT mode indeed, will fix it asap.
Thanks for the report!

erwincoumans, 2020-05-05:
Fixed here:
https://github.com/bulletphysics/bullet3/pull/2781/commits/d479ca712ce4dac3357486d53ae692d6f39ab9a3
Will be in pypi soon

closed 2020-05-05 by erwincoumans