bulletphysics/bullet3 #2782: Segfault in changeVisualShape while using eglRenderer

colin-r-schultz, 2020-05-05:
I have found a segfault that occurs consistently under the following conditions:

1. The eglRenderer plugin is loaded
2. A body is removed
3. A body is created (likely using the same id as the previously removed body)
4. `changeVisualShape` is called on the newly created object

The following code shows the simplest case where I could consistently reproduce this. 
According to `faulthandler`, the segfault occurs at `changeVisualShape`.
```
import pybullet as p
import pkgutil
import pybullet_data

p.connect(p.DIRECT)
egl = pkgutil.get_loader('eglRenderer')
plugin = p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")
print("plugin=", plugin)

p.setAdditionalSearchPath(pybullet_data.getDataPath())
obj1 = p.loadURDF("plane.urdf")
p.removeBody(obj1)
obj2 = p.loadURDF("plane.urdf")
p.changeVisualShape(obj2, -1, rgbaColor=[1, 0, 0, 1])

p.unloadPlugin(plugin)
p.disconnect()
```
My uneducated guess is that this occurs because the removed body and new body share the same id. I'm wondering if, as a workaround, there is a way to stop id pooling and force each new body to have a new id?

erwincoumans, 2020-05-07:
Thanks for the report, will check it out.

BlGene, 2020-09-06:
@colin-r-schultz have a look if this works for you: https://github.com/BlGene/bullet3/tree/egl_remove_fix

erwincoumans, 2020-11-13:
I think this should be fixed in my recent commit:
https://github.com/bulletphysics/bullet3/commit/1bd82e7d81d3a265563538b7882eef80f215f0e8

Please reply/re-open if not.

closed 2020-11-13 by erwincoumans