bulletphysics/bullet3 #2786: GCC Error - LIB\bullet3-2.89\btgui No such file or directory

loopahnoob, 2020-05-07:
I'm honestly so confused as to how to install this. I've tried using the python method, however that needs the VC compiler, which I don't have. I'm thinking that maybe MinGW isn't compatible with Bullet. Anyways, here's the log:

[1_5-7-2020.log](https://github.com/bulletphysics/bullet3/files/4590384/1_5-7-2020.log)

erwincoumans, 2020-05-07:
Bullet should compile out-of-the-box on Linux, Mac OSX and Windows using a recent free Visual Studio version.
If you prefer a custom compiler, it is up to you to make the build work.
btw, the log file is partial and shows no error.

closed 2020-05-07 by erwincoumans

loopahnoob, 2020-05-07:
Is it possible to use MinGW though? I don't have Visual Studio or any IDE installed, I only use a text editor and the MinGW compiler. If this is not possible, then I will consider installing VS.