bulletphysics/bullet3 #2803: How to

GiZumRyue, 2020-05-14:


erwincoumans, 2020-05-14:
Please see https://github.com/bulletphysics/bullet3/blob/ef08e9b415f05d4fed60f3846730d9579962b61f/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py

GiZumRyue, 2020-05-14:
> Please see https://github.com/bulletphysics/bullet3/blob/ef08e9b415f05d4fed60f3846730d9579962b61f/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py

Thx! I have found out the solution!

closed 2020-05-14 by GiZumRyue

wagenaartje, 2021-01-30:
To others that want to know what the actual reason is you get this error: a joint can have multiple states. So let's say you have an object with 3 hinge joints, which you want to assign the angles `[-pi,+pi,-pi]`. Then don't do:

```
pybullet.resetJointStatesMultiDof(objectId, [0, 1,2], [-np.pi, np.pi, -np.pi])
```

But do

```
pybullet.resetJointStatesMultiDof(objectId, [0, 1,2], [[-np.pi], [np.pi], [-np.pi]])
```

In other words, add an extra dimension.