bulletphysics/bullet3 #2808: Question: Is there any reason to not build Bullet3 with OpenMP multi-thereading?

Jacalz, 2020-05-17:
Hello there. I am looking at improving the package of bullet3 in the Solus Linux Distribution and I saw that there is support for multi-threading. I took a peek at Fedora and Arch, but neither has multi-threading enabled as part of the packages. Thus I am wondering if there is any compelling reason from a packaging standpoint to not build multi-threading using OpenMP?

This might be a stupid question as I don't now the API, but will multi-threading entablement benefit already built code utilizing Bullet3 (as in using it with Godot games for example)?

ArneDJ, 2020-05-19:
I have a semi-related issue.
I am using Linux too and I can't even find a way to build Bullet3 with multithreading enabled.
I tried looking for an option in the cmake files but I only found multithreading options for Bullet 2.
None of the examples in the example browser seems to have anything related to multithreading either.

If someone could tell me how to build Bullet3 with multithreading I would be very thankful.

erwincoumans, 2020-05-21:
multi-threading  is a user-contribution that I never tried/tested, and I'm not planning to maintain it. If you want to enable it, good luck! Also, Bullet3/OpenCL is not active for many years now, probably best to remove the Bullet3* folders.

closed 2020-05-21 by erwincoumans