bulletphysics/bullet3 #2810: PyBullet: Programmatically set joints position and limit with createMultiBody

ycps, 2020-05-18:
Hi, I'm experimenting with randomized robot generation, but could not find a way to set joints position or limit with `createMultiBody`: joints seem to always be created at the center of shapes without range limits.

I understand that using URDF or the other importers is the expected way to create robots, but I don't know what is the best approach when most of the robot features should be randomized.

I searched about this, but could only find this similar questions in the PyBullet forum:
- [Programmatically set the revolute (hinge) joint limits?](https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=24&t=12927&sid=00c3f4ea457563f36e1e3883eeaa48cb)
- [how to change joint position](https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=11953)

Thank you for Bullet!

erwincoumans, 2020-05-18:
Joint position can be set using the linkPositions argument.
See this example:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/createMultiBodyLinks.py

Joint limits are indeed not exposed yet.

erwincoumans, 2020-07-08:
Here is a CL that allows to enable/modify lower/upper joint limits and also joint limit max force.
See https://github.com/bulletphysics/bullet3/pull/2919

closed 2021-04-26 by erwincoumans