bulletphysics/bullet3 #2817: Collision algorithm for area check

AndreaCatania, 2020-05-21:
@erwincoumans in godot engine we are using the gjk/epa algorithm to detect area overlaps.

That algorithm is really slow/non-optimal for such use. What is the optimal algorithm that bullet provides, or one that you can advice to use, for areas overlap detection? Is the SAT already available in bullet?

Thanks!

erwincoumans, 2020-05-21:
What is the exact shape of the area? If it is an axis aligned bounding box, you can use the broadphase checks. Otherwise, it depends on the shape.

AndreaCatania, 2020-05-21:
I need to support all the convex shapes, and I would just need an array of bodies that overlaps. I was thinking about SAT algorithm.

erwincoumans, 2020-05-21:
Yes, Bullet has the SAT implementation, but it needs to be organized to get what you need.

For the convex shapes, you need to call initializePolyhedralFeatures, so SAT can be used.
To try out SAT, you can set dispatchInfo.m_enableSAT = true.

The code is here:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionDispatch/btConvexConvexAlgorithm.cpp#L489

You probably only need
foundSepAxis = btPolyhedralContactClipping::findSeparatingAxis

It would probably be better to implement a dedicated area check using this:
gather all objects that have AABB overlap (using a broadphase query)
for those potential colliders, use this SAT check.

AndreaCatania, 2020-05-22:
Perfect, thanks for these info, I'll integrate it and let you know.

AndreaCatania, 2020-10-05:
The function `foundSepAxis = btPolyhedralContactClipping::findSeparatingAxis` is really good for what I've to do, though it uses `btConvexPolyhedron` and not regular shapes.

I'm interested to know why such chose. It's much optimal the curren algorithm compared to the (eventual one) that uses primitive shapes?

I'm asking because this function forces me to also maintain a list of shapes, which is not so nice.

closed 2021-04-26 by erwincoumans