bulletphysics/bullet3 #2821: DeepMimic Gym env queries actions at different frequency from the paper

ManifoldFR, 2020-05-24:
The DeepMimic paper specifies that the policy is queried at 30Hz. This is ensured in the initial TensorFlow code that was copied to this repository: when the `update()` method of the RLAgent class is called, it invokes a [method](https://github.com/bulletphysics/bullet3/blob/5233b72160df8e8915320b03793d85dd0c6792ba/examples/pybullet/gym/pybullet_envs/deep_mimic/env/pybullet_deep_mimic_env.py#L220) called `need_new_action()`, which compares the internal environment time to a variable called `needs_update_time` which is increased by 1/30 seconds when a new action is applied.

However the gym environment `step()` method [here](https://github.com/bulletphysics/bullet3/blob/5233b72160df8e8915320b03793d85dd0c6792ba/examples/pybullet/gym/pybullet_envs/deep_mimic/gym_env/deep_mimic_env.py#L104) does none of that, and only steps the simulation once by calling the update method of the [PyBulletDeepMimicEnv class](https://github.com/bulletphysics/bullet3/blob/5233b72160df8e8915320b03793d85dd0c6792ba/examples/pybullet/gym/pybullet_envs/deep_mimic/env/pybullet_deep_mimic_env.py#L262) to apply the action directly: this results in the policy being queried at 240Hz in the Gym env.

I have a fix for this which would change the behaviour of the Gym env to make it more consistent with the paper and implementation.

erwincoumans, 2020-06-02:
Thanks for the help. The DeepMimic gym env hasn't been tested yet, it would be great to get the Gym environment to train/enjoy. So far, I've been using the DeepMimic_Optimizer.py to train and testrl to enjoy, both don't use the gym environment.

erwincoumans, 2020-06-02:
applied PR, thanks!

closed 2020-06-02 by erwincoumans