bulletphysics/bullet3 #2823: Bullet Physics ExampleBrowser using openGL3+ [btgl] Release Build (Not Responding)

jgarc248, 2020-05-25:
Hello, I'm fairly new to Pybullet and just trying to learn as I go along. 
When I first started using Pybullet, I really didn't have my issues with crashing or anything similar. The Example Browser ran just fine without any issues. However, lately, whenever I go to use p.connect(p.GUI), the window for the example browser shows up but it gets stuck in the "Not Responding" state. I tried looking if there any similar issues but couldn't find any.

I'm running this on an Intel i7 @2.2 GHz, 16 GB of RAM, and GeForce GTX 1060 

This is the code that I'm trying to run. I'm not sure if my code is an issue because until a few days ago it would run and work fine. 

`import pybullet as p 
import time
import pybullet_data
import pybullet_envs

physicsClient = p.connect(p.GUI)

p.resetSimulation()
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
planeId = p.loadURDF("plane.urdf")
p.setGravity(0,0,-10)

pendulum1Id = p.loadURDF("pendulum.urdf", [0,0,1], [0,0,0,1], 0, 1, 0, 3.0)

p.getNumJoints(pendulum1Id)
#p.getJointInfo(pendulum1Id, 0)
#p.getJointState(pendulum1Id, 0)

maxForce = .001 
mode = p.VELOCITY_CONTROL
p.setJointMotorControl2(pendulum1Id, 0,
 	controlMode=mode, force=maxForce)

p.setRealTimeSimulation(1)`

erwincoumans, 2020-05-25:
What OS?

jgarc248, 2020-05-25:
Hello, my OS is Windows. 
I just tried your solution and everything is working. 
I really appreciate your help! Have a nice day!

closed 2020-05-25 by jgarc248

erwincoumans, 2020-05-25:
Sure. Temporary disabling rendering makes loading much faster, but after that you need to re-enable it otherwise you don't see anything.