bulletphysics/bullet3 #2827: How to get the joints' world Position?

GiZumRyue, 2020-05-27:
Hello!
The function getJointsState() is to get the jointPosition (as jonit angle or joint orientation quaternion).
But I want to get the joint's world positions xyz.
(I know the getLinksState() return the positions of the middle of the links which is not what i want.)
Is there any way to deal with it?
Any help? Thanks a lot!

ShravanTata, 2020-05-27:
Hi @GiZumRyue,

getLinkState returns a tuple consisting of multiple return values, of which worldLinkFramePosition/Orientation is what you are looking for.

`(joint_position, joint_orientation) = pybullet.getLinkState(bodyUniqueId, linkId)[4:6]`

GiZumRyue, 2020-05-27:
> Hi @GiZumRyue,
> 
> getLinkState returns a tuple consisting of multiple return values, of which worldLinkFramePosition/Orientation is what you are looking for.
> 
> `(joint_position, joint_orientation) = pybullet.getLinkState(bodyUniqueId, linkId)[4:6]`
> 
> PS : You should be using [pybullet-forum](https://pybullet.org/Bullet/phpBB3/) for questions like these.

Awesome! It's This! And thanks for ur advise!

closed 2020-05-27 by GiZumRyue