bulletphysics/bullet3 #2831: Issue with motor joint drift

Teslatic, 2020-05-29:
I have an URDF file that constructs a rotating crane on tracks.

**Problem description**
The gantry of the crane can be rotated around the origin by giving force on the respective motor joint "hub_to_gantry". In the attached example file, the joint is accelerated up to a certain number of simulation cycles and after hitting that threshold it decelerates down to a applied force of 0.

There are two versions of the crane available.
a) The hoist of the crane is stiff and cannot sway in any direction
b) The hoist is connected to the trolley via two hinges in order to make it swayable in 2 dimensions.

**Expected behaviour**
Gantry accelerates, decelerates and then stays in position.

**Observed behaviour**
Gantry accelerates, decelerates and does not stay in position because of a drift. The whole gantry continues moving due to this drift and despite the fact, that no force is applied and friction is enabled.

**How to recreate the behaviour**
For the swayable crane: `python main_bullet_issue.py`
For the non-swayable crane: `python main_bullet_issue.py --nosway`

**Used versions**
Python: 3.7.7
Pybullet: 2.7.8

Any help is appreciated.

[drift_issue.zip](https://github.com/bulletphysics/bullet3/files/4701005/drift_issue.zip)

erwincoumans, 2020-06-02:
I don't notice a drift and with or without --nosway the hoist is not moving. The yellow crane is modeled as a single convex shape (press 'w'), it is best to model it using convex decomposition (either using v-hacd or manually).

Teslatic, 2020-06-02:
Thank you for your model improvement suggestions.

Regarding the drift, I can still observe it, even with the better model.

A video is attached, you can see the drift when you watch the mouse pointer in the end of the video. 
[drift_issue_vid.mp4.zip](https://github.com/bulletphysics/bullet3/files/4715535/drift_issue_vid.mp4.zip)

Steven89Liu, 2020-06-05:
hi Teslatic, do you have any progress?

Teslatic, 2020-06-05:
No, I do not have any progress. Over the course of weeks (before I opened this issue) I have tried many different modifications of the URDF file, but none resulted in success.

Steven89Liu, 2020-06-05:
i think the drift is caused by the joint of tracks_to_gantry, this issue can be avoided by changing the joint type to fixed.

Teslatic, 2020-06-08:
When I change the joint tracks_to_gantry to "fixed", a new problem with the tines arises:
I rotate them along with the gantry in the same fashion. When the "tracks_to_gantry" joint is continuous, the tines close properly. When it is fixed, the tines continue revolting, despite being limited by the URDF setting.

The gantry drift seems to be gone though.

Do you have an explanation for this behaviour? (See attached vides for demonstration, where only the joint type has changed).

[tines_limit_issue.zip](https://github.com/bulletphysics/bullet3/files/4746026/tines_limit_issue.zip)

Steven89Liu, 2020-06-09:
what i see is different from yours as below,  and i observe that your speed of tines is -1 so i guess maybe you have other change. i will check this later supposing you have set the desired speed of tines to -1.
```
Cnt: 164 Sending command {'grapple': 0.0, 'hoist': 0.0, 'gantry': 1.6500000000000012, 'trolley': 0.0, 'tines': 0.0}
velocity gantry= 0.06502565557683673 trolley= 0.0 rope= 0.0 grapple= 0 jam= 0
hub angle: 0.0
Gantry angle: 0.18013498316872956
Trolley pos: 3.211243908196362e-05
Sway angle X: -1.1125799252679049e-13
Sway angle Y: -1.4355460299973502e-06
Hoist pos: 1.6479873021779688e-18
Grapple rotation angle: 8.264603304641822e-07
Tines angle: -0.00015537920006926487
```

Steven89Liu, 2020-06-10:
by default, the infoGlobal.m_splitImpulse is enable and use the btMultiBodyConstraintSolver which seems that it doesn't handle the m_rhsPenetration.  meanwhile i have a doubt that if it can meet the requirement of the velocity and limit constriants at the same time.

Steven89Liu, 2020-06-14:
i still think it is very hard to meet these requirements of the velocity and limit constraints at the same time with the mode VELOCITY_CONTROL which depends on the order that one constraint being processed later will have more priority. for this case, i add two minor changes as below to avoid this drift.
```
p.setPhysicsEngineParameter(enableFileCaching=0, numSolverIterations=10)

p.setJointMotorControlArray(
                self.crane_body_id,                                                                                                                                                                                                          
                [
                    self._joint_map["tracks_to_gantry"],
                    self._joint_map["hub_to_gantry"],
                    self._joint_map["gantry_to_trolley"],
                    self._joint_map["sway_y_to_hoist"],
                    self._joint_map["grapple_base_to_grapple_body"],
                    self._joint_map["grapple_body_to_left_grapple_jaw"],
                    self._joint_map["grapple_body_to_right_grapple_jaw"],
                ],
                controlMode=p.VELOCITY_CONTROL,
                targetVelocities=[
                    0,
                    gantry_velocity,
                    trolley_velocity,
                    rope_velocity,
                    grapple_rot_velocity,
                    -jaw_force,
                    jaw_force,
                ],
            )
```

Teslatic, 2020-06-15:
Thank you very much for your suggestions, they indeed seem to eliminate the drift issue with the gantry. 

However, the Tines are still moving beyond their limits. They are supposed to stop at their respective limits. Any idea, what that could be or is your previous answer related to this problem as well.

Should I file another separate issue for the limit issue?

Steven89Liu, 2020-06-16:
i think it is better to file another separate issue.

Steven89Liu, 2020-06-17:
you can try my last change which also can avoid the over limits but it is not a common solution that need we to manually adjust the parameters.

Teslatic, 2020-06-17:
Closing, for the drift issue seems to be resolved. Thank you for your help.

closed 2020-06-17 by Teslatic