bulletphysics/bullet3 #2833: Can't instantiate two Humanoid environments at the same time (second one gets stuck)

rdednl, 2020-06-01:
As title, when instantiating more than one environment, the second environment gets stuck and returns always the same state.

Note: 
- this doesn't happen for any other environment other than Humanoid
- this was not happening 10 days ago (we have reliable results on those date)

Versions: 
gym = 0.17.2
pybullet = 2.7.9

Minimal working example:

```
import pybullet
import pybullet_envs
import gym
import numpy as np

env_name = 'HumanoidBulletEnv-v0'

env_1 = gym.make(env_name)
env_2 = gym.make(env_name)

env_1.reset()

equal_state = env_2.reset()

for i in range(100):
    state, reward, terminal, _ = env_2.step(env_2.action_space.sample())
    if not(np.array_equal(equal_state, state)):
        print("not equal")

print("all states equal")
```

erwincoumans, 2020-06-01:
Thanks for the report! If true that's a bad regression. If you have more details information what PR caused it, it could be resolved quicker.

rdednl, 2020-06-01:
I tried to look my history and it doesn't seem I updated nor pybullet nor gym recently (I was on 2.7.4, but updated to 2.7.9 to see if it would have been solved there) and I am not sure what could have caused it to appear now.

I managed to fix it in #2835 (at least, so it seems)

closed 2020-06-01 by rdednl